slic3r/Slic3r #3836: Enforce Supports not functioning

jaggzh, 2017-04-03:
Fresh build on April 2, 2017: 1.2.9-901-g90623c4
OS: Debian Stable

Description:
"Support material -> Enforce support for the first N layers" doesn't seem to do anything.
I have Supports disabled, then any number entered in the Enforce option.

If I'm understanding correctly, this would allow me to have supports generated for N layers, and the rest of my object (above that) be support-less.

lordofhyphens, 2017-04-15:
Test case:
[3836.zip](https://github.com/alexrj/Slic3r/files/923953/3836.zip)

jaggzh, 2017-04-15:
Disable "Don't support bridges", or use a model without vertical sides below the Enforce layer count and it should work.

My .zip is not uploading for some reason (Something went really wrong, and we can't process that file. Try again.)

PR #3877 (2017-04-27): Addressed logic of enforced supports. 
There are inconsistent rare crashes which I cannot find as related to my changes.

Fixes #3836
closed 2017-04-27 by alexrj