slic3r/Slic3r #3839: Slic3r no longer slices in background, gives div by zero error when trying to export g-code

lordofhyphens, 2017-04-03:
Reported by @mrehorst in #3837 as a second issue, moved here to separate.

### Version
1.2.9-890-gbc35063a

### Operating system type + version
Win 7 64 bit SP1

### Behavior
I uninstalled Strawberry Perl, deleted old slic3r, deleted old boost, and followed sequence in the wiki to reinstall everything (git clone, etc.).  Building slic3r fails test 92.  Slic3r opens fine, when I load any stl file the object appears on the plater but does not start slicing (yes, background slicing is turned on).  

#### STL/Config (.ZIP) where problem occurs
[Desktop.zip](https://github.com/alexrj/Slic3r/files/888552/Desktop.zip)

lordofhyphens, 2017-04-03:
I can't reproduce this with a current fresh build from http://dl.slic3r.org/dev (1.3.0-dev-90623c4) with Windows 10.

alexrj, 2017-05-01:
I can't reproduce this anymore.

closed 2017-05-01 by alexrj