slic3r/Slic3r #3840: Request to remove tooltip timeout feature

ruggb, 2017-04-03:
### Version
 ==1.2.9 stable
### Operating system type + version
 ==Windows 10 1607 14393-953
### Behavior
tooltip timeout is insufficient for me to get thru more than 1/2 of the tip.
then I must move mouse to recover tip - once read, I usually must reread to fully comprehend which means more mouse movements.

_Is this a new feature request?_ YES, or maybe a less of a feature request
OK, its me and I could get a brain replacement but maybe you can just **eliminate the timeout feature for tooltips** and just make it respond to hover. Is there really a need to time out the tip?

lordofhyphens, 2017-04-03:
The timeout value on 1.3.0-dev (windows) is around 30 seconds. This is currently the maximum length allowed due to an upstream (wxperl) bug. This can't be fixed until the entire UI has been replaced. 

[current development version](https://bintray.com/lordofhyphens/Slic3r/slic3r_dev/_latestVersion#files) (currently an alias to http://dl.slic3r.org/dev).

ruggb, 2017-04-03:
thks 30 sec works much better. I guess I am testing now.

lordofhyphens, 2017-04-03:
I didn't make any change at all.

ruggb, 2017-04-03:
I understand - I dnld 1.3.0 and that is what I was referring to. It loaded and integrated with Repetier-host fine on my desktop, but somehow I didn't do the same thing on my laptop cause it doesn't work yet.

ruggb, 2017-04-03:
found it - thx

closed 2017-04-03 by ruggb