slic3r/Slic3r #3842: Excessive supports generated, I think.

jaggzh, 2017-04-03:
Built on April 2, 2017: 1.2.9-901-g90623c4
OS: Debian Stable

Description:
The attached object (image also attached) seems to get supports generated excessively.  The supports run up the sides where no support is needed (45 degree limit on supports).  The model seems intact, although it does have small faces and some sharp edges/corners.

At least it would be nice to know what qualities of the object result in this behavior.  Thanks a lot.

Ideally: Only what seem to be true overhangs would be supported (near the very bottom, and perhaps up to the circle hole... but not the sides and such).

[foo1](https://cloud.githubusercontent.com/assets/20318973/24604815/3b3bea3c-181b-11e7-9099-b56236373496.jpg)
[test-switch-slot.zip](https://github.com/alexrj/Slic3r/files/889658/test-switch-slot.zip)

lordofhyphens, 2017-04-03:
Change support threshold to 150% or 200% (tweak that value until you get what you are interested in)

Closing.

jaggzh, 2017-04-03:
Even at 0 threshold it engulfs the object.  (Front-most object in set here)
[image]

jaggzh, 2017-04-04:
Even at 1 it builds supports all over the object's surface.

lordofhyphens, 2017-04-06:
@jaggzh Please zip and add the config you were using.

lordofhyphens, 2017-04-06:
[image]

Here's a picture of one of my results and the model rotated 90 in Y
[image]
[0.05config.zip](https://github.com/alexrj/Slic3r/files/902664/0.05config.zip)

It would help as well if you included the object in the orientation you are seeing this issue in. 

Config from #3860, object rotated 90 deg in Y.
[image]

Same config, changed to 60 degrees.
[image]

Arakon, 2017-04-09:
I agree with this issue. Comparing my usual "filament test print object" between 1.3.0 and 1.2.9, 1.3.0 generates way, way more support material than 1.2.9. Keep in mind that with the lower amount of support material, the object printed perfectly fine, i.e. there were no badly supported areas.
[image]
[image]

lordofhyphens, 2017-04-09:
Please post your configuration, this is related to that (and of course
update to newest, I am now enforcing the new configuration option
semantics.

jaggzh, 2017-04-10:
[supp0rts.zip](https://github.com/alexrj/Slic3r/files/911613/supp0rts.zip)

jaggzh, 2017-04-10:
[image]

[image]

These are the orientations.

lordofhyphens, 2017-04-11:
I think I found the bug, I needed an extra pair of () around an if statement in SupportMaterial.pm.
[image]

commit 5cc2d346 (2017-04-11): Wrong logic on detecting use of % for support threshold. Fixes #3842.
closed 2017-04-11 by lordofhyphens