slic3r/Slic3r #3843: Background processing is active all the time

Tinchus2009, 2017-04-03:
### Version

1.2.9-901-g90623c4

### Operating system type + version
Win 10 64b

### Behavior
Background processing is disabled in my preferen ces. Despite of that, when I change values in the printing setting, as soon as I close tthe windows it starts the slicing in the background
Im thinking that may be this is the intended behaviour? because the processing only starts when closeing the window.

* _Steps needed to reproduce the problem_
Load a model, open the printing settings wndow, change for example vertical shells value (no processing when you changed the value), close the printing settings window, the processing starts here.

lordofhyphens, 2017-04-03:
#3839

alexrj, 2017-04-04:
Fixed, thank you!

closed 2017-04-04 by alexrj