slic3r/Slic3r #3844: This does not slice as expected, is it me or slic3r?

ruggb, 2017-04-03:
### Version
1.2.9 stable

### Operating system type + version
windows 10 1607

### Behavior
this model has one wall = 1mm and the opposite wall = 2mm. If I slice it with infill the walls are 1 or 2 (as set) perimeters wide and the whole model is infilled. If sliced without infill the walls are 1 or 2 (as set) perimeters wide with no infill. Is it possible to slice this and get 1 and 2 mm walls (with infill where necessary) and a hollow core? Does the model need to be different or Slic3r settings different? Or is this not possible with Slic3r?

0.2mm layers, 1 or 2 perimeters 15% infill

[one wall.zip](https://github.com/alexrj/Slic3r/files/889957/one.wall.zip)

_Is this a new feature request?_NO

lordofhyphens, 2017-04-03:
http://dl.slic3r.org/dev

xoan, 2017-04-03:
Are you sure you attached the correct model? If the answer is _yes_, sorry, I can't understand what do you are trying to do... :\

[image]

ruggb, 2017-04-03:
Yes, the model has one wall that is 1mm thick and the opposite is 2mm. It is drawn with one box inside the other and positioned slightly offset to achieve the above. The bottom & top are 2mm thick.
When I slice it, Slic3r gives me the frame with the set # of perimeters and the rest of it is infill instead of being hollow and just infilling the walls. I haven't tried putting a hole in it, which might make Slic3r realize the inside is hollow or it might think it is not manifold.

lordofhyphens, 2017-04-03:
Please add a few screenshots showing the issue.

ruggb, 2017-04-03:
this is an xray view. The right wall is 2mm, the left is 1mm, the bottom is 2mm,
the slice (1 perimeter, 15% infill) does not implement the internal hollow. It is all infilled.
the other slice is with 0 infill and again there is no internal structure.
If I put a hole into the hollow it will slice as expected.
This is done with 1.3.0 - I didn't try the hole on 1.2.9, but the others do the same thing with 1.2.9
[image]
[image]
[image]
[image]
[image]

lordofhyphens, 2017-04-03:
So the original 3d model has no internal voids?

xoan, 2017-04-03:
I think your model has the internal faces inverted, so it is computed as a whole solid shell.

ruggb, 2017-04-03:
we are getting into semantic confusion.
The orig internal structure is closed with 2mm between it and the external structure except for one wall which is 1mm. Obviously, if I want it to be hollow I must open it to the outside air with at least one hole.
At this point, the question is, why do I need to put air into it to get Slic3r to make it hollow as in the 5th & 6th pic? Is that the way it has to be, or is Slic3r missing something?

xoan, 2017-04-03:
I guess it is because I said in my previous comment: the faces in the internal shell are inverted when there is no connetion between internal and external shells of the object.

Probably, your CAD software (SketchUp?) solve this issue when you make a hole connecting the shells. So, the problem is with your CAD software. Try by searching how can you reverse the normals for the inner shell without doing a hole.

ruggb, 2017-04-03:
Well now, that is just amazing. When I put a hole into the model Sketchup automatically reversed those faces. Since it is virtually impossible to see that, I never thought of it. Of course, I probably never would have even if I could see it since I have no clue how Slic3r does things. It is all magic to me.
I reversed all the faces in the original and it works.
thank you for that education.

ruggb, 2017-04-03:
thx again

closed 2017-04-03 by ruggb

lordofhyphens, 2017-04-03:
Sketchup is pretty bad when it comes to producing useful models/STLs. I strongly suggest avoid using it.

ruggb, 2017-04-03:
Everybody says that, but I have mostly figured it out and it is relatively easy for me. There are a few quirks as with any program. But I keep looking at other things to see how much of a learning curve there would be.
So far, I am happy with Sketchup.