slic3r/Slic3r #3846: Tooltip for Filament Settings/Cooling/Disable fan

ruggb, 2017-04-03:
### Version
1.3.0 dev

### Operating system type + version
Windows 10 1607

### Behavior
Tooltip for Filament Settings/Cooling/Disable fan for the first...________...layers
reads "You can set this to a positive value to disable fan at all during the first layers, so that is does not make adhesion worse. (default 3)"

Might be better worded.........
"This disables the fan completely for the first # of layers to aid in the adhesion of media to the bed. (default 3)"

lordofhyphens, 2017-04-03:
The text lives in PrintConfig.cpp, make a pull request ;)

Please check out the i18n branch and also edit the corresponding msgid in all the .po files.

ruggb, 2017-04-03:
You are making a big assumption that I know what I am doing!
But I will be happy to look and see if I can figure it out.

lordofhyphens, 2017-04-03:
Yep, this is pretty easy (hint: checkout the i18n branch, and do a text
search for the string you want to replace).

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

ruggb, 2017-04-05:
OK,these changes are committed in my fork ruggb/Slic3r. There is one for each file. patch 1, 2, & 3.

I guess that is the way it works. Now what?????????

in printconfig.cpp @ 169
//    def->tooltip = __TRANS("You can set this to a positive value to disable fan at all during the first layers, so that it does not make adhesion worse.");
169       def->tooltip = __TRANS("This disables the fan completely for the first # of layers to aid in the adhesion of media to the bed. (default 3)");

..............................
in en_US.po @ 289
      //**"You can set this to a positive value to disable fan at all during the first "
      //**"layers, so that it does not make adhesion worse."
289   "This disables the fan completely for the first # of layers "
290   "to aid in the adhesion of media to the bed. (default 3)");

and @ 291
//**"You can set this to a positive value to disable fan at all during the first "
//**"layers, so that it does not make adhesion worse."
292   "This disables the fan completely for the first # of layers "
293   "to aid in the adhesion of media to the bed. (default 3)");

xoan, 2017-04-05:
https://help.github.com/articles/fork-a-repo/#next-steps

lordofhyphens, 2017-04-05:
1) Create a branch
2) make sure branch has your work in it
3) make a github pull request (in this case, the first entry in the list is
i18n branch, the second is your branch).

lordofhyphens, 2017-04-05:
4) Add "Fixes #3846" to the pull request description

ruggb, 2017-04-05:
OK I think I have done all the damage I can do. I don't know where #38**5**6 came from above.

alexrj, 2017-05-01:
Until i18n is not merged, people would need to do two PRs for each change... maybe we can ignore i18n for now, and update it just before merging?

PR #3856 (2017-04-05): I18n - fix for #3846
this fixes #3846 - this is my very first attempt so check it out.......