slic3r/Slic3r #3849: [Feature Request] Load config from gcode file

lordofhyphens, 2017-04-04:
This has been asked for a few times over IRC -- the ability to load a GCODE export that Slic3r produced and have it be treated like a config.ini file. It would be convenient for people who routinely keep exports around for backup purposes. 

I also know people tend to upload gcode output from Slic3r in their bug reports, this would obviate the need to have a second config.ini export (which apparently only two out three people can do reliably).

This is generally possible because Slic3r already saves the primary print settings as comments at the end of the file. 

From my observation, the last blank line in a file signifies the start of the configuration.

It would definitely be slower-loading than a regular config.ini import.
The enhancement would consist of processing an input file and looking for the last newline in the file and processing from there (alternatively, we could check every line for the ; XXXX = YYYY and let the configuration validator deal with the rest).

ProfDrYoMan, 2017-04-05:
You could use a comment in the first line of the exported G-code to store the byte position where the slic3r configuration lines start.
Then it would be easy to skip() there and not have to read the complete and big C-code file.

nerocide, 2017-04-05:
Few lines config lines at the begining af a gcode file should also be added.
@ProfDrYoMan Otherwise does something like GNU head/tail in pure perl can be used? Something that can trig a comment containing an equal sign.

ProfDrYoMan, 2017-04-05:
Repetier firmware uses an approach for their firmware config.
See https://www.repetier.com/firmware/dev/index.php on the download tab.
Configuration is written into a comment as json at the end of the file.

lordofhyphens, 2017-04-05:
Slic3r already dumps it's config at the end of files and I would like for this to be backwards compatible. 

The way that it writes gcode makes it difficult to insert a reference at the top to something below.

drhender, 2017-05-29:
I believe this is a duplicate of #3150.

lordofhyphens, 2018-02-23:
Thanks @foreachthing for the pull request.

closed 2018-02-23 by lordofhyphens