slic3r/Slic3r #3851: Confusion about extrusion WIDTH vs extrusion SPACING

djdelorie, 2017-04-04:
### Version
1.2.9-920-g139aeb8
### Operating system type + version
Fedora 24 64-bit

### Behavior
In a recent design, I wanted a decorative top layer using concentric fill.  I specified the width of the part to be a multiple of extrusion width, and... there was a big gap in the middle.  After digging through the code, I discovered that the extrusions are squished together to account for the rounded edges.  Ok, makes sense in retrospect... but it's very difficult to design a properly-sized infill when you have no way of knowing what the actual extrusion spacing is going to be (the information is hidden deep inside the code, and depends on the layer height too!) and for certain decorative objects, I don't want the extrusions squished together like that.

I don't know the right way to fix this, but it took me quite a bit of head-scratching to figure out what was going on, and I'm still not sure I got my math right.  Some suggestions:

* change the priority between width and spacing so that flow is adjusted to compensate for spacing, instead of the other way around

* document (i.e. in the GUI as other changes happen) what the inter-extrusion spacing would be, so at least the designer knows

* adjust infill extrusion width for decorative patterns like concentric so that it's always a whole number of extrusions

* allow the user to override the OVERLAP internal setting, at least for top/bottom infill, preferably for perimeters and other infill too.  Note that the default 100% overlap setting means you have to get your extrusion multiplier EXACT or you'll overextrude.  A default of 95-99% might make more sense.

alexrj, 2017-04-04:
Following up to your points:

1. It's not true that spacing is adjusted to compensate for width. The spacing formula is constant.
2. The math is explained in the [manual](http://manual.slic3r.org/). Live-calculating the spacing is not totally easy but doable. Still, this needs to be supported by good test cases.
3. Width is already adjusted/strecthed for having whole number of extrusions in rectilinear and concentric infill -- unless there's a bug or the shape is not convex.
4. Show a test case for this.

Slic3r has been around since 2011, and flow math is quite mature as it inspired other software out there. There's been a large amount of discussion within the community on these topics; you'll find pages here on GitHub and elsewhere. We really can't change things if there's no strong evidence that they need to be changed. And generally speaking, every request has to be supported by discussion of actual test cases.

djdelorie, 2017-04-05:
[spacing-bug.zip](https://github.com/alexrj/Slic3r/files/894820/spacing-bug.zip)

I agree that the math makes sense, my point here is that the intended behavior can be surprising if you're not aware of it.  Sample included.  The part is 2.4mm wide with 0.4mm extrusions, and the expected 6 extrusions don't fill the whole space.  I *understand* why it happens, but I didn't *expect* it to happen, and how to make it "magically" work is not obvious (I assume you have to disable the extrusion width override, but then you've lost control of the appearance of the decorative layer).  The fact that the math is described at the bottom of a web page under "advanced topics" - instead of, say, in the tool tip for the extrusion width - means many folks (including me, and I've *read* the web page before) won't get a chance to learn about it before they discover it's not working the way they expect.

Like I said, I don't know the right way to fix this.  I suppose either changing slic3r to do what the user expects, or changing the user's expectations, would be equally good ;-)

Tinchus2009, 2017-04-05:
Well... I think you are asking for fixing something actually doesnt need t be fixed man...
The formula for path generation is out there for quit a long time and it is the same formula used by other slicers as well and it has proven to work really well. What you are looking for it is a way to have perfect looking finishes and for that... you dont use a slicer, you use post processing.
And I desagreewith you:  what the users want is a slicers that simplifies their job, almost none of the normal users wants to elar or hear about maths, formulas, etcs.
If your printer is properly calibrated, and you use a manual extrusion value and other tweaks, you can get perfect surface finishes.

MyPrinter7, 2018-02-14:
Hi All,

I have a query relating to this. I am currently using Slic3r to create paths for a Bioprinter. 

The problem I am having is that for the purposes of biomaterial printing we need the infill's "inter-extrusion" spacing to be known, consistent and controllable. Is it possible with Slic3r to set an infill spacing of for example 1.5mm?

Would this only be possible by modifying the source and compiling Slic3r?

Any help would be super appreciated - It would be great to know that this is possible before diving in :)

Thanks!

lordofhyphens, 2018-02-14:
@Myprinter7 you will need to look at the different Fill C++ source files in xs/src/libslic3r.

We have defined spacing for support, but what you are asking for is not something that has come up yet.

MyPrinter7, 2018-02-14:
Thanks for the speedy response and for the pointer hyphens. 

I would like also to put this forward as a feature suggestion. I know multiple scientists in my area alone all bashing their head against the wall with this. The lattices produced especially in tissue engineering need to be highly consistent, particularly when aiming to mimic the native tissue mechanical character/porosity and other qualities.

For now I will have a look as you have advised and report back with any findings so that others may have the answer. Any additional help is most appreciated.

Have a great day.

powertide, 2018-03-07:
@MyPrinter7  Hi~I'm now looking for the same thing as you did. I want the line spacing  between extrusions in rectilinear infill to be known and consistent. 
Actually I have tried to do a few modifications, overriding the "line_spacing" variable in FillRectilinear.cpp but failed to compile. 
Have you made any progress? It'd be really nice to have someone to study this together.