slic3r/Slic3r #3853: Confusion about new settings-in-dialogs design

djdelorie, 2017-04-04:
### Version
1.2.9-920-g139aeb8

### Operating system type + version
Fedora 24 64-bit

### Behavior
Some comments about the new settings-in-dialogs design (vs the old settings-as-tabs design)...

When I'm setting up a print, I tend to tweak the settings and look at the 3D print for a while, until it prints what I want.  Which means I used to constantly switch between plater and settings tabs, which was OK, but with the new system you see both the plater and the settings at the same time.  This sets the user's expectations to be "I can change the settings and see what the effect is".  Not true, even with background slicing enabled.  It wasn't obvious that the view is regenerated when you close the dialog.  Further, each time I re-opened the dialog, it switched back to the first "page" of settings, instead of remembering where I was.  Also, the main window becomes unresponsive when a settings dialog is open, without giving any visual indication that it's so.

Some ideas for addressing this: 

* If you show the 3D view while settings can be changed, if background processing is enabled, the user expects that the view will update as they change settings.

* There should be an obvious way to force a reslice, preferably without closing the settings dialog so you don't lose your place.

* Allow the user to interact with the main window (at least, the 3D view pan/tilt) while a settings dialog is open

* Remember where the user was in each settings dialog, so when you re-open it after forcing a reslice, at least you don't have to also change pages each time.

* allow multiple settings windows to be open simultaneously

* Since the plater is the only tab, there's no point taking up screen area with the "Plater" tab.

alexrj, 2017-05-20:
Thank you for this valuable feedback. It's almost a duplicate of #3835, or very close to it, but it contains useful ideas.

lordofhyphens, 2017-06-04:
1. We now have option of tabs or dialogs (and the Controller tab), so we do need to take that room up. 
1. It's automatic with background processing on, switching back to the layers tab if it is off. Both are available in the new scheme of things.
1. You can have the windows open now and still interact with the plater.
[image]

closed 2017-06-04 by lordofhyphens