slic3r/Slic3r #3859: Dont suport bridges is failing big

Tinchus2009, 2017-04-06:
### Version
1.2.9-944-g04b59af

### Operating system type + version
_win10 64b

### Behavior
supporting bridges was failing some time ago, but with the latest build something has changes and I think the problem is bigger now. Big and clear bridges are generating supports.
I have included config files and 2 screenshots.
[falla2.zip](https://github.com/alexrj/Slic3r/files/901233/falla2.zip)

Tinchus2009, 2017-04-06:
This is a more simple but clear example: a common bridge test. I have printed this test with previous version without problems, I remember support was not geenrated. The model is http://www.thingiverse.com/thing:113160

And with supports at 120% (degault) you get fully supported bridges
[image]

lordofhyphens, 2017-04-16:
@Tinchus2009 With your test case, it looks like this regression has gone as far back as 1.2.0. 

1.1.7 does not seem to exhibit this problem.

Tinchus2009, 2017-04-16:
To be sincere i have not gone back that far. There were reports of supports
being created under bridges when gapfill was active, deactivateing gapfill
solved the problem somehow (actually is not a solution since you loose the
gapfill functionality). But now the problem got worse because all bridges,
even those not useing gapfill are being supported. 1 or 2 weeks ago,  after
i builted a dev version i issued a git log command to see what changed and
i remembered seeing thata new logic to support generation was added. It was
then that i tested it suppirt generation and saw this behaviour.
Nevetheless i can say the problem appeared with that change.
You jave already confirmed that in 1.1.7 there is no problem or you would
like me to also test that version?

lordofhyphens, 2017-04-16:
I tested it in 1.1.7 and 1.2.0 with the old packaged builds. I can't build the differences between the two though with my newer Perl so bisecting is out unless I want to deal with replacing the entire toolchain.

Tinchus2009, 2017-04-16:
auch... you are right. testing older versions is also out fr me haveing the
new perl

alexrj, 2017-05-01:
Fixed. Thank you for reporting this!

closed 2017-05-01 by alexrj