slic3r/Slic3r #3861: Support Pattern Angle not respected

foreachthing, 2017-04-06:
### Version
slic3r-master.2017.04.05.885.66c7eb7

### Operating system type + version
Windows 7 Pro

### Behavior
* Added support material - Pattern: Rectilinear; spacing: 2.5 mm
* Angle at 0 degrees, all good
[image]

* Angle at 90 degrees ... so I thought!
[image]

In the second screenshot the infill pattern is definitely not at 90°.

The pattern jumps around, when using the angles 29 to 31 ...

* 29 degrees:
[image]
* 30 degress:
[image]
* 31 degrees:
[image]

After that, I stopped messing around and used 30 degrees - close enough, Come to think of it, I'll rotate the part!

alexrj, 2017-04-11:
Thank you for the report.

Also, interesting seeing how many options you exposed in the right panel.

foreachthing, 2017-04-12:
> how many options you exposed in the right panel. 

Just the ones I change from print to print :)
The layout gets slow, though!

lordofhyphens, 2017-04-15:
Fill takes its angle in radians, SupportMaterial was providing it in degrees. Fixed it by applying deg2rad in all invocations of set_angle() in SupportMaterial.pm  
[image]

[image]

commit 1857bf63 (2017-04-15): SupportMaterial operates in angle, underling XS Fill operates in radians. Convert degrees->radians
Fixes #3861
closed 2017-04-15 by lordofhyphens