slic3r/Slic3r #3862: Background processing don't trigger when new object is added

gege2b, 2017-04-06:
Hi all

There are some issue about background processing opened, but no one with these symptoms
my appologies if this is a duplicate 

### Version
04b59af (current build from bintray)

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### Behavior
when adding a new object, the background processing didn't trigger. 
If i'm on the preview tab when adding the object, it is emptied first (seems logic) but that's all

I need to change a whatever setting to trigger the background processing (as expected)

* _Steps needed to reproduce the problem_

- Open slic3r with background processing enabled
- Load an object
- Nothing is done
- Change a setting and see the BP begin

* _Expected Results_

Background processing to trigger when adding an object

* _Actual Results_

Nothing happens

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

config here : 
[slic3r_config.zip](https://github.com/alexrj/Slic3r/files/903063/slic3r_config.zip)

STL is not important

lordofhyphens, 2017-04-07:
@gege2b just as a fyi we've migrated away from bintray, http://dl.slic3r.org/dev has everything new and it's sorted by date/time.

lordofhyphens, 2017-04-07:
Well, I can't reproduce with those steps on Linux or on my Windows 10 system with the default config or the provided configuration. 
04b59af

1. Started Slic3r.
1. Ensured background processing was on.
1. Loaded 1 STL. It started slicing. 
1. Loaded a different STL. It started slicing.

I changed no additional configuration options.

foreachthing, 2017-04-07:
I can reproduce this on my computer with Windows 7 Pro and slic3r-master.2017.04.05.885.66c7eb7 (1.3.0-dev).

1. Start Slic3r (Background processing is turned on).
2. Import STL.
3. .... nothing! Slice3r tells me: Loaded modelname.stl and just sits there until I change a setting.

By only clicking a rotate button, Slic3r starts slicing.

gege2b, 2017-04-07:
@lordofhyphens yep I know, I said bintray but I was thinking of the dev from the slic3r website, my bad

I tested right now on my Kubuntu 16.04 with a freshly pulled repo ( ab278f03e17873dc423e011b49da4a480fb806fd ) and I can reproduce it again with the same steps and default config

xoan, 2017-04-07:
I can reproduce it too.

current master (ab278f03e17873dc423e011b49da4a480fb806fd) and also precompiled binaries from http://dl.slic3r.org/dev/linux/ in Arch Linux.

lordofhyphens, 2017-04-07:
I can't, which means that I can't debug it.

FWIW I've never liked this feature.

gege2b, 2017-04-10:
OK I understand
I have reactivated the background processing because when disabled, every change to a setting switch the tab from preview to 3D plater witch is a bit annoying (and as long as I remember, it seems to be the expected behaviour)

I can try to locate where is the problem but can you please help me to know where to look first ?
I.E. witch function is triggerred when a new object is added on the plater OR/AND when a setting is changed ? it would save me some time

RafaelEstevamReis, 2017-04-10:
In plater.pm there are a function called start_background_processif I
remember well.
You can search for what calls it in the same file.

(https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm)

(I do not touch this file for a long time, so if someone else tells you to
do something else, go for them)

gege2b, 2017-04-10:
Thank you !

I think I've found the culprit, seems to be a single missing line. 

I Will submit a PR and let @lordofhyphens or @alexrj decide what to do

alexrj, 2017-04-27:
I think I fixed this with some recent commit, and I cannot reproduce it. @gege2b, do you have a chance to test and confirm? Thank you!

gege2b, 2017-04-27:
unfortunately, I just tested on linux mint and I still get this issue

xoan, 2017-04-27:
I still get this issue in my system too (Arch Linux)

And it is fixed by applying #3873

lordofhyphens, 2017-04-27:
@xoan I see the incorrect behavior with that patch applied (and do not on current master).

As @alexrj said in the PR thread that pr doesn't call anything that isn't called before.

There is a weird race condition going on.

xoan, 2017-04-27:
@lordofhyphens yes, I know the patch it has no sense, but when applied, background processing is triggered when adding an object in my system with current master.

Could it be related to some weird threading behavior? I made a simple debug in `resume_background_process`:

```perl
sub resume_background_process {
    print "resuming... ";
    my ($self) = @_;
    
    if ($self->{process_thread} || $self->{export_thread}) {
        Slic3r::resume_all_threads();
        print "1\r\n";
    } else { print "0\r\n"; }
}
```
and `1` is never printed :\

xoan, 2017-04-28:
OK, probably I'm taking a stab in the dark, but commenting out [Plater.pm#L1504](https://github.com/alexrj/Slic3r/blob/master/lib/Slic3r/GUI/Plater.pm#L1504) makes background processing to be triggered when adding an object.

Version: current master (1.2.9-972-g9bdc537c) (9bdc537c)
System: Arch Linux

gege2b, 2017-04-28:
hi there
I confirm that commenting this line makes the background processing to trigger as expected under linux mint

alexrj, 2017-05-01:
I'll need to test on Linux then...

alexrj, 2017-05-02:
This can be reproduced on any platform by enabling "Auto-center parts".

alexrj, 2017-05-02:
Fixed! Can you confirm?

xoan, 2017-05-02:
I can confirm it is fixed in Arch Linux :)

gege2b, 2017-05-02:
no surprise, it's fixed also in linux mint :)
great job

alexrj, 2017-05-03:
Great, thank you!

closed 2017-05-03 by alexrj