slic3r/Slic3r #3867: XS/C++ compile issue(GUI.cpp bugs)

powertide, 2017-04-07:
I followed exactly the instructions on https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows.

I'm using Windows7 64bit.
Strawberryperl 5.24.0 64-bit
boost 1_63_0

It went well till when I did "perl Build.PL", and it looked like this

[image]

There seems to be some problems in the file "GUI.cpp". 
Really looking forward to your help.

lordofhyphens, 2017-04-07:
Are you using the appropriate version of Strawberry Perl?

There have been changes, see wiki.

lordofhyphens, 2017-04-07:
So here's the build log from the last build on the server:
https://ci.appveyor.com/project/lordofhyphens/slic3r/build/1.3.0-master-926

That build (and the numerous ones before it) pass, so I'm not particularly persuaded that there is a problem in GUI.cpp (https://github.com/alexrj/Slic3r/blob/master/xs/src/slic3r/GUI/GUI.cpp). It's been unchanged for over a year.

SetThreadExecutionState is a Windows API function, so it should be in Windows.h
https://msdn.microsoft.com/en-us/library/windows/desktop/aa373208(v=vs.85).aspx

https://github.com/alexrj/Slic3r/blob/master/appveyor.yml is what our build server does on every build. You can examine the individual files it calls (the .ps1 scripts in package/win) and alter paths to suit your local system.

lordofhyphens, 2017-04-07:
Related:
http://stackoverflow.com/questions/25276092/c-win-api-function-not-declared-in-this-scope

We use mingw-w64's compiler in our setup, are you sure that you don't have another MinGW install in your path before C:\Strawberry\c\bin?

powertide, 2017-04-10:
@lordofhyphens Thanks a lot for your help. I did have several other MinGW installed in my computer before, and after I had them deleted, the problem I proposed didn't occur again. But another one came up.
[image]
Already tried setting the BOOST_DIR variable to "C:\dev\boost_1_63_0", didn't help. The output stayed the same.
I am sure that I successfully installed boost in C:\dev\boost_1_63_0 before and compiled it(used command bootstrap and b2 as instructed in the Wiki page). 
And I sure looked it up in the issues, but couldn't find a valid answer .

lordofhyphens, 2017-04-10:
That's not the right version of gcc (sjlj exceptions WILL hang slic3r during the test phase). If you were using the correct mingw as referenced by the wiki for x86_64 (on 64-bit you MUST use Slic3r Perl), the gcc version would be 6.3.0.

The "can't find boost" failure is not necessarily because there is no Boost in that directory, it's that the Boost found can't be used. You need to have Boost installed for every compiler you build with. 

Here's a copy of boost built with the appropriate compiler (gcc 6.3.0), which is what our build server uses. 
https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=boost_1_63_0-x64-gcc-6.3.0-seh.7z

powertide, 2017-04-10:
[image]
As you can see ,the gcc 4.9.2(silj) seems to be just the gcc in strawberryperl directory. The strawberryperl version I installed was 5.24.0 x64. Could it be the problem? But it was exactly the version the wiki page recommended. Hell I'm confused...

lordofhyphens, 2017-04-10:
https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#slic3r-perl

The wiki changed about a week or so ago. I know because I was the one to change it.

If you reread the preceding section you will see that note to not use the 64 bit version from strawberryperl.com.

If there is another section of that page that told you to use 64 bit strawberry perl 5.24 tell me and I will fix it (the previous recommended 64 bit version was 5.18 and that version won't work anymore ).

powertide, 2017-04-11:
OK. I thought that the "32-bit only" meant for the computer used. So I should install 32-bit strawberryperl even though my computer is 64-bit, right?  I'll give it a go. Thanks~

lordofhyphens, 2017-04-11:
That is almost the *opposite* of what I said. Download and install [Slic3r Perl](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows#slic3r-perl), which is what the wiki says to do if you are using Windows 64bit.

You may need to overwrite C:\Strawberry with [this archive](https://bintray.com/lordofhyphens/Slic3r/download_file?file_path=Strawberry-6.3.0-seg-archive.7z)

powertide, 2017-04-11:
First of all, thank you for your patience. @lordofhyphens 
But I installed the custom strawberryperl , recompiled boost and did as you/wiki said, and the problem is still there.
[image]

The gcc version shown now has changed to 6.3.0. That is some progress I guess.

powertide, 2017-04-13:
I really appreciate your help @lordofhyphens, and I have completed the "Perl Build.PL" as below following your advice.
[image]
But where can I find the output of this build( executable file)? 
I can't find it in Slic3r directory , nor in Strawberry.
Could you show me please?

lordofhyphens, 2017-04-13:
See package/win directory if you want to build something that looks like an exe.

If you just want to start Slic3r, run the Perl program with slic3r.Pl as an argument.

lordofhyphens, 2017-04-13:
Reference the deployment part of appveyor.yml for hints

powertide, 2017-04-13:
[image]
No exe file was found in package/win directory.
And what did you mean  by "the deployment part of appveyor.yml"? 
Could you be more specific, please?
Plus, good to see that the slic3r program was successfully started by running"slic3r.PL".

powertide, 2017-04-13:
[image]
Found only one "appveyor.yml" file on my computer, and it looks like this.
The deployment part seems to not exist. What am I missing here?

powertide, 2017-04-13:
I've done some search and found several issues regarding： #1777  #2316
I tried using PP (pp -o slic3r.exe slic3r.pl)  and did get a slic3r.exe, but it didn't run as I thought. A black window just flashed a sec and disappeared, nothing else happened.
[image]
@lordofhyphens @alexrj I know you guys are very busy, but really hope you can show me how to package slic3r into an executable file. It's very important to me.

alexrj, 2017-04-13:
@powertide, the packaging script is package/win/package_win32.ps1

Please note we have automated fresh builds at http://slic3r.org/download so you don't need to package yourself.

lordofhyphens, 2017-04-13:
You need to compile the exe (which is just a thin wrapper to invoke the
Perl interpreter on Slic3r.pl). Unless you really really want an exec file
it isn't strictly necessary.

I mentioned appveyor.yml because that contains the instructions our build
server uses to produce what goes to dl.slic3r.org/dev/win.

The script name should be compile_wrapper.ps1 and you will need to run it
with the argument "524"

lordofhyphens, 2017-04-13:
If you want an archive to copy to another PC, package_win32.ps1 will do
that.

lordofhyphens, 2017-04-13:
````
deploy_script:
- ps:  "& package/win/appveyor_deploy.ps1"
````

Read that script and see what it calls.

closed 2017-04-13 by lordofhyphens

powertide, 2017-04-14:
@lordofhyphens  @alexrj  Yes, I do need to get an exec file so that it can be used on other computers without perl environment.
I tried running compile_wrapper.ps1 and it turned out like this:
[image]
From what I can tell, the errors displayed seem to be grammar mistakes. But since the compile_wrapper has been used so many times, I believe there are some pieces I'm missing here.  Anyways I will let you decide. The file looks like this:
[image]

lordofhyphens, 2017-04-14:
Open up powershell as a prompt and run from there. Cd to the directory (it
is better than cmd.exe imho).

Then run
& compile_wrapper.ps1 524

Or read the file and use the commands one by one, accounting for any
variables.

powertide, 2017-04-17:
1.  I did as  @lordofhyphens   said but the result stayed the same.
[image]
From what I can tell, the error says that it lacks the terminator “ ' ”.
I'm a newbie on this, please take a minute to look into this for me.

2. Then I used **the commands one by one**, and the result was weird.
[image]
Nothing happened. No mistake , and certainly no files generated after this.

lordofhyphens, 2017-04-17:
Remove the | Write-Host parts from commands.

lordofhyphens, 2017-04-17:
Is your default locale non-US (like shift-JIS)?

powertide, 2017-04-17:
@lordofhyphens Yes, I'm in China.

lordofhyphens, 2017-04-17:
Read my earlier message  (I edited it).

lordofhyphens, 2017-04-17:
As an aside,.would you be willing/able to contribute to a Chinese translation of Slic3r? Mechanically making it work is very easy.

powertide, 2017-04-17:
Yes, I can do translation, and I'm glad to help contribute to slic3r community~I will be so honored!

powertide, 2017-04-17:
Well, I erased the "| Write-Host" parts, but it didn't make any difference.
I don't think the alternative you @lordofhyphens suggested about downloading packages directly would work for me.  
Because I want to open the setting for " **infill phase**", which means increasing a certain value of infill angle by each layer. For example, if the infill angle for layer 1 is 90°, and the infill phase is 45°. Then for layer 2 the infill angle would be 90+45=135°. And layer 3 would be 135+45=180°. ...
As far as I know, the infill phase in slic3r for now seems to be fixed and cannot be changed. 
(Maybe it should be put forward in an feature request, since I don't find any related request.)
So yes, I want to alter this, generate my own ".exe" files and copy to another PC to use them.

lordofhyphens, 2017-04-17:
Slic3r.exe does not have any functional part of Slic3r in it (this is
different than 1.2.9). It is basically a simplified form of perl.exe that
calls slic3r.pl.

So yes, you can drop it and all dll files in the root directory of a
packaged Slic3r dir into another source dir and it will work (assuming you
have built libslic3r accordingly via Build.pl).

If you installed slic3r-perl as the wiki indicated, g++ will be in your
PATH.

You can move shell.cpp into the same directory and compile it directly if
you want to keep trying to build the wrapper exe.

There isn't much else I can do to help without having access to your
computer to figure out what is different on your system.

lordofhyphens, 2017-04-17:
The old packing script just copied perl.exe and perl524.dll from
C:\strawberry to the slic3r directory  (it did a few other things with PAR,
glhf with that if you are having problems getting the compile wrapper
script to run).

You can cheat though by copying your entire slic3r source directory  (where
you built slic3r) into an extracted copy from the build server (overwriting
everything). It is a dirty hack, but it would give you a working,
self-contained archive.

FYI there are known issues with paths that contain non-latin characters on
Windows. I am still trying to find a working solution for that.
On Apr 16, 2017 10:39 PM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> Slic3r.exe does not have any functional part of Slic3r in it (this is
> different than 1.2.9). It is basically a simplified form of perl.exe that
> calls slic3r.pl.
>
> So yes, you can drop it and all dll [...]

powertide, 2017-04-17:
Thank you for your always patient explaination and kind help.
Here is what I found when I tried the "dirty hack" method you mentioned.(Desperate time, desperate measures~ can't blame me :-) ):

[image]
This is a package I downloaded and extracted. The "duobianxing.stl" file was manually added into the directory. As you can see, the directory path has none no-latin character.
[image]
And when I entered command "slic3r-console duobianxing.stl", errors came out. It has something to do with the locale codeset. I looked into the "locale.pm" file:
[image]
But meanwhile, it's perfectly OK to run "perl slic3r.pl", the program started in secs.
So, how should I modify the "locale.pm" file to make the exe work?

lordofhyphens, 2017-04-17:
Okay, that is some kind of bug. I suspect the file encoding of the source
needs to be fixed. It is probably why you can't run the scripts either.

Very weird, I will look into it today.

powertide, 2017-04-19:
Have you got any clues on how to fix this bug? @lordofhyphens

lordofhyphens, 2017-04-19:
yeah, save the encoding as something that isn't cp932 (which is what I think it used to be because of my editor config somehow). 

Resave the files as utf8 and it should *probably* work.
I also just checked in updated versions of the affected files.

powertide, 2017-04-19:
I saved the file "Locale.pm" as utf-8 and tried running the command again. But it turned out the same.
Do I need to change other files too? @lordofhyphens

lordofhyphens, 2017-04-19:
@powertide just update, the problem isn't in locale.pm it was in in the compiled exe. 

Get a new windows build.

powertide, 2017-04-19:
OK I'll try it.

powertide, 2017-04-19:
Downloaded the latest windows build.
[image]
It didn't work. All the same again...

powertide, 2017-04-20:
The latest window builds don't work on my computer(win7/X64). 
I think this might be high priority.　@lordofhyphens 　How can I help?

powertide, 2017-04-24:
Any progress on this problem here @lordofhyphens ?
Hope you have not forgotten about it :-)

lordofhyphens, 2017-04-24:
I have a day job, it is really busy right now. I am also basically out of ideas.

powertide, 2017-05-10:
@lordofhyphens Just so you know,  the slic3r exe from the latest window builds does work fine now on my PC. I am not sure since from which version it has been working fine though. 
Still,  good news to know! :-)

lordofhyphens, 2017-05-10:
Well, so long as it's working now. I don't have any real way of debugging this any further. If it stops working for you then open a new ticket.

powertide, 2017-05-10:
Sure~ Thank you for your always hard work!

DaveAM, 2017-08-25:
I had the same problem and know nothing about powershell, but noticed in the powershell editor that the formatting changed for the initial '. So the following worked for me (note I'm using Win7). I simply put a space after all the -L and -I and before the "C:\strawberry..." (e.g. -I 'C:\strawberry\perl\lib\CORE\').

powertide, 2017-09-21:
@DaveAM  WOW I did as you said and it has amazingly worked!
Thank you so much. Though I still don't get it. What led you to edit the ps1 file like that?