slic3r/Slic3r #3868: Extreme sudden (temporary) slowdown halfway through print

Hertog99, 2017-04-07:
### Version
1.2.9-945-gab278f03 (ab278f03e17873dc423e011b49da4a480fb806fd)

### Operating system type + version
Ubuntu 16.10
### Behavior
Halfway through a print with  all speed settings set to 'auto' and Max print speed set to 60, the printer suddenly starts to run very slow for infill and perimeters, travels are still fast. The slowdowns last for a couple of layers, and after that, return back to normal. In the attached STL file, the problems start at layer 56 and are over by layer 63. Speeds go as low as 0.8 mm/sec 

The speeds can be seen in http://gcode.ws, so seem printer unrelated.
* _Steps needed to reproduce the problem_
See STL file at layer 56 to 63
* _Expected Results_
Somewhat faster prints between layer 56 and 63

Either this is functionality that I just don't understand yet, or it is the result of having everything on auto, and have the system default to some very low speeds.

#### STL/Config (.ZIP) where problem occurs
Attached are copy of config.ini and the gcode as generated by slic3r.
[speed-problem-slic3r.zip](https://github.com/alexrj/Slic3r/files/906676/speed-problem-slic3r.zip)

lordofhyphens, 2017-04-07:
Could you attach a zip archive of the STL file?

Hertog99, 2017-04-07:
STL-file as requested.

[Y_Axis_motor_mount_v2.STL.zip](https://github.com/alexrj/Slic3r/files/906758/Y_Axis_motor_mount_v2.STL.zip)

lordofhyphens, 2017-04-07:
Looks like autospeed is in use, which is probably near root cause of the problem.

Hertog99, 2017-04-07:
All settings are indeed on 'auto'.

[image]

lordofhyphens, 2017-04-07:
Yeah, there's probably an area in that part that is being extruded slowly and that rippled out the rest of the layers (autospeed is about maintaining pressure).

Thank you for the test case.

alexrj, 2017-05-04:
The first thing I notice is that you're using a 0.35mm extrusion width with a 0.4mm nozzle. That doesn't really look like a sane configuration.

alexrj, 2017-05-04:
It was related to that very narrow extrusion width, and I fixed this.

closed 2017-05-04 by alexrj

Hertog99, 2017-05-05:
Thanks for looking into it. I probably overdid the 'set extrusion width slightly smaller than nozzle diameter' -tip I saw somewhere.