slic3r/Slic3r #3870: Wx-0.9928 Fails to Build

NewtMan, 2017-04-08:
### Version
Building from source, origin/master branch up-to-date from a pull today

git rev-parse HEAD yields:
30076331bd85b556935c22bd518a5e7cf4bb40e3

### Operating system type + version
Linux monet 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1+deb8u2 (2017-03-07) x86_64 GNU/Linux
gcc version 4.9.2 (Debian 4.9.2-10)

### Behavior
perl Build.PL succeeds
Perl Build.PL --gui fails building Wx-0.9928

Excerpt from the associated build.log file:
g++ -pthread   -c -I. -I.. -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/lib/wx/include/gtk3-unicode-3.0 -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/include/wx-3.0  -I. -I. -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/lib/wx/include/gtk3-unicode-3.0 -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/include/wx-3.0   -I. -I../.. -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/lib/wx/include/gtk3-unicode-3.0 -I/usr/local/lib/x86_64-linux-gnu/perl/5.20.2/Alien/wxWidgets/gtk_3_0_2_uni/include/wx-3.0  -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g   -DVERSION=\"0.01\" -DXS_VERSION=\"0.01\" -fPIC "-I/usr/lib/x86_64-linux-gnu/perl/5.20/CORE" -DWXPL_EXT -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__   -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__    -DWXPL_EXT -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__   PropertyGrid.c
PropertyGrid.c: In function ‘void XS_Wx__ColourProperty_SetPlValue(PerlInterpreter*, CV*)’:
PropertyGrid.c:27130:49: error: call of overloaded ‘wxVariant(wxColour&)’ is ambiguous
                 wxVariant value = wxVariant(*col);
                                                 ^
Seems the compiler is choking.  Interesting thing is that I've successfully built Slic3r from source before and have been running it for months.  This error cropped up after the latest git pull.  Any ideas?

lordofhyphens, 2017-04-08:
I've taken to editing ext/propgrid/PropertyGrid.xsp and replacing that call to wxVariant to wxAny.

Tinchus2009, 2017-04-11:
I can report that the issue is also present in the win10 64b bulids

lordofhyphens, 2017-04-11:
@Tinchus2009 It's an upstream issue, not something we can fix (and we are NOT maintaining wxperl ourselves). 

You are *strongly* encouraged to use my slic3r dependency archive on Windows.

Tinchus2009, 2017-04-11:
Im already useing that package. What I did was unzipped it to the slic3r directory. Should I do anything else?
Also, I read that the perl build.pl --gui command is not necesary to run  it i every build?

lordofhyphens, 2017-04-11:
@Tinchus2009 If you're using my dependency archive, `perl Build.PL --gui` is completely unnecessary.

@NewtMan Apparently there was another release of wxperl yesterday: 0.9929

lordofhyphens, 2017-04-11:
Wxperl 0.9929 released April 9 resolves issue on Linux.

closed 2017-04-11 by lordofhyphens

NewtMan, 2017-04-11:
Unfortunately, Wx 0.9929 does not solve the problem for me.  I get the same error building PropertyGrid.c

lordofhyphens, 2017-04-11:
... I just downloaded the archive from CPAN on my Debian Stable system and
built it.

Perl 5.20

NewtMan, 2017-04-20:
So, taking your lead to try "from scratch", I kicked off a VirtualBox VM, installed a fresh copy of the latest Debian distro and built Slic3r successfully, including Wx.  Guess there's something botched up in my desktop installation - time to re-install :-(

lordofhyphens, 2017-04-20:
I also just built on Debian Stretch (testing) and that looks fine too.