slic3r/Slic3r #3872: Pan in View - Model disappears

foreachthing, 2017-04-10:
### Version
slic3r-master.2017.04.05.885.66c7eb7

### Operating system type + version
Windows 7 Pro

### Behavior
* Load any model
  * wait for Slic3r to finish slicing (change to any tab you like: 3D, 2D, ...)
* change view to **front**, **left**, **back** or **right**
  * Pan with right mouse button
    * my model disappears!
* change view to **top** or **bottom**
  * Pan with right mouse button
    * view is panned as expected

lordofhyphens, 2017-04-15:
Looks like some kind of asymtope thing going on with that particular camera vector. I suspect what is going on is that it's pulling the entire plate behind the camera (or otherwise exaggerating the moves).

lordofhyphens, 2017-04-15:
Yep, the intercept plane ends up with a huge -Y coordinate if you are viewing from the front, and a huge +Y from the back.
`Cursor plane: (-37.8130463854417,-752964704.704026,0)`

With left/right it's X of course.

lordofhyphens, 2017-04-15:
Here's the output from dumping the event GetX/GetY and the  calculated line and intercept plane:

Diagnonal "default" angle:
````
Cursor pos: (120,180)
Mouse line start/end: (-1650.72367771136,-1182.6789224324), (2774.27665757898,3242.32141285793)
Cursor plane: (-113.619218325621,354.42553695333,0)
Cursor pos: (125,179)
Mouse line start/end: (-1649.7311141594,-1183.15296225649), (2775.26922113094,3241.84737303385)
Cursor plane: (-112.367393386566,354.210758516346,0)
Cursor pos: (129,178)
Mouse line start/end: (-1650.35702605922,-1183.04557312311), (2774.64330923112,3241.95476216723)
Cursor plane: (-112.734043899285,354.577409036827,0)
Cursor pos: (133,178)
Mouse line start/end: (-1648.89042215544,-1184.51217702689), (2776.1099131349,3240.48815826345)
Cursor plane: (-111.267439995502,353.110805133045,0)
Cursor pos: (133,177)
Mouse line start/end: (-1652.08289115075,-1181.83823079801), (2772.91744413959,3243.16210449232)
Cursor plane: (-114.200647603713,356.044012749018,0)
Cursor pos: (134,177)
Mouse line start/end: (-1652.23476342195,-1182.72340502111), (2772.76557186838,3242.27693026922)
Cursor plane: (-114.352519874921,355.158838525918,0)
````
Front:
````
Cursor pos: (465,170)
Mouse line start/end: (-22.8913068604543,-2950.00000459638), (-22.8913068604543,5900.00013848313)
Cursor plane: (-22.8913068604543,-593917495.270956,0)
Cursor pos: (466,170)
Mouse line start/end: (-22.8913068573703,-2950.00000459638), (-22.8913068573703,5900.00013848313)
Cursor plane: (-22.8913068573703,-593917495.270956,0)
Cursor pos: (467,171)
Mouse line start/end: (-22.8913068542863,-2950.00000459267), (-22.8913068542863,5900.00013848684)
Cursor plane: (-22.8913068542863,-591977895.155919,0)
Cursor pos: (468,172)
Mouse line start/end: (-22.8913068586528,-1942550.12500486), (-22.8913068586528,-1933700.12486178)
Cursor plane: (-22.8913068586528,-591977894.850588,0)
Cursor pos: (469,171)
Mouse line start/end: (-22.8913068481182,-3882150.25000487), (-22.8913068481182,-3873300.24986179)
Cursor plane: (-22.8913068481182,-595857094.775329,0)
Cursor pos: (470,171)
Mouse line start/end: (-22.891306867386,-1942550.12500487), (-22.891306867386,-1933700.12486179)
Cursor plane: (-22.891306867386,-593917494.965625,0)
Cursor pos: (470,172)
Mouse line start/end: (-22.891306867386,-1942550.12500486), (-22.891306867386,-1933700.12486178)
Cursor plane: (-22.891306867386,-591977894.850588,0)
````

Looking at it more carefully, it looks like the orig point is being calculated with what looks like some absurd -Y value.

````
Cursor pos: (451,220)
Mouse line start/end: (-24.0782634135783,-2950.00000441108), (-24.0782634135783,5900.00013866843)
Cursor plane: (-24.0782634135783,-496937489.519085,0)
Origin: (-24.1630460245157,-496937489.519085,0)
````
Figured out why, too: 
````intersect_plane 
z:       0.000000e+00 
a.z:     2.256952e+01 
b.z:     2.256991e+01 
b.z-a.z: 3.868458e-04
````
floating point is generating tiny Z values for the a.z-b.z that translate to a large value in the reciprocal.  

So in these corner cases the intersect plane is not being calculated properly, probably because of floating point rounding. 

I'm not sure what to do here other than perhaps add some special processing to round very small values from the subtraction to 1 (we don't want to round to 0 because of divide-by-zero). 

Any thoughts @alexrj ?