slic3r/Slic3r #3874: Solid infill threshold area eliminates all infill

Tinchus2009, 2017-04-11:
### Version
1.2.9-947-gef8695d

### Operating system type + version
win10 64b

### Behavior
Im posting this issue useing a stl as model, but I have confirmed this bug with other models as well.
I was useing the "solid infill threshold area" option to get rid off the infill of small areas of a model. 

[image]

Since the defaul value of 70 was big, I reduced it to 40 mm2, and I got the desired results:

[image]

But then I noticed that all infill in the model had desapeared.

[image]

I layed a littel changeing some values but it didnt worked. No mattter what value you use, removeing infill areas useing this option will remove the infill in the hole model, even in the areas that are bigger than the infill threshold area value.

[falla4.zip](https://github.com/alexrj/Slic3r/files/911876/falla4.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

foreachthing, 2017-04-11:
Works in 1.3.0-dev (_slic3r-master.2017.04.11.932.5cc2d34_), with or without your settings, as expected.

Tinchus2009, 2017-04-11:
Well, It doesnt worked . I have builted that new dev version and still is not working, at least under win10, and a strange thing: in order to get no infill in those small areas I had to reduce the no infill threshold area value to 10, but doing so removes all the infill of the object also.

lordofhyphens, 2017-04-11:
@Tinchus2009 have you grabbed a download from dl.slic3r.org/dev/win

Tinchus2009, 2017-04-11:
No. I builted it as always: git pull, perl build.pl
It builted ok

Tinchus2009, 2017-04-11:
Ok, I have discovered the problem... Useing the version I downloaded from the server also didnt worked.
What Im seeing is that: in my desktop shortcut to slic3r, I use --datadir F:\slicer_config so slic3r uses values stored in a secundary disk I have. Somehting there is makeing the new build to nor properly work. If I delete that parameter from my shortcut, slic3r uses default datadit location, and then I get the expected results.
So there is a problem with the parameter  --datadir F:\slicer_config ????

xoan, 2017-04-11:
`--datadir` is not a parameter for Slic3r, so probably it is something related to the operating system itself (something like working directory in *nix systems)

lordofhyphens, 2017-04-11:
@xoan yes it is a parameter in Slic3r: https://github.com/alexrj/Slic3r/blob/master/slic3r.pl#L39

@Tinchus2009 The default location for Slic3r profiles on Windows is somewhere in your user folder (local settings, etc, it's a pain for me to remember where it is anymore).

`F:\slicer_config` is a real path that exists, right?
I can't tell what is going on here.

xoan, 2017-04-11:
@lordofhyphens sorry, I didn't found it in README so I supposed it wasn't from Slic3r. My fault.

lordofhyphens, 2017-04-11:
Yeah, the README.md can lag behind the options list in slic3r itself.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Tinchus2009, 2017-04-11:
Ok, I got it. The problem actually comes from a different place:
I think I read a similar report about this, cant find it right now. What I did is: I tested again changeing the solid infill threshold area values. I changed it twice, infill generation was ok, till I changed it the third time. The values used were: first time I changed it to 15: solid infill in those small holes was generated, and the hole infill of the object was ok. Second change was 10, still solid infill was generated for the small areas (I knew this was going to happend) and the rest of the object infill was ok too. The third change: I set a value of 7, and this time no infill was generated in the small areas but neither in the hole object.
I saved the configuration... closed slic3r , re opened it, loaded the stl again, and went to the preview, no change in aprameter at all. This time the small areas did not had infill and the object infill was ok.
I repeated the process a couple of times: the problem is not that option, the problem is the background sliceing. When you change values in the printing settings, most of the times when you close the tab slic3r go back to the normal platter and you can have a new slice clicking in the preview tab. But some times a change in the printing settings does not trigger this behaviour. Slic3r dont go back to the 3d platter, and if you click in the preview tab, no new sliceing is triggered.

lordofhyphens, 2017-04-11:
Good catch. Hopefully this can be reproduced.

lordofhyphens, 2017-04-11:
Does using that option as an override show the same behavior?

Tinchus2009, 2017-04-11:
Yes

bcolsen, 2017-04-12:
I have this as well on Ubuntu with latest master.

To Reproduce:
1: Add an stl file
2: Override bottom or top solid layer number
3: slic3r will not recompute the infill and the model must be deleted and reimported

[image]

Tinchus2009, 2017-04-18:
this issue is solved. i have tsted a couple of time. super great work

closed 2017-04-18 by Tinchus2009