slic3r/Slic3r #3875: [Feature Request] Scale Modifiers

foreachthing, 2017-04-11:
### Version
Any

### Operating system type + version
Windows 7 Pro

### Behavior
#### Current Behavior
* Add cylinder (box, slab, stl, ...) modifier
  * make some changes to the model and reload it
  * notice, modifier is suddenly too small or too big
    * or the modifier was chosen just a bit too small
  * need to **delete** modifier and **redo all modifications** (first you need to take notes or make a screenshot of current settings)

#### NEW Behavior
* Add cylinder (box, slab, stl, ...) modifier
  * make some changes to the model and reload it
  * notice, modifier is suddenly too small or too big
  * scale modifier a bit in X and some more in Z - done.

_Is this a new feature request?_
Yes.

lordofhyphens, 2017-04-11:
Is a right-click menu similar to how it works on the Plater (right-click on the modifier in the menu in this case) sufficient?

alexrj, 2017-04-11:
It's a good start :-)

lordofhyphens, 2017-04-12:
Did some looking over the code, and I think we'd be well-served by factoring out the scaling subroutines in Plater.pm so we can call them from ObjectSettingsDialog.pm.

foreachthing, 2017-04-12:
I thought it could be like the `$optgroup_movers` except scale it in X, Y or Z.
Maybe limit the scale factor to be less than 5 times platter size ;)

alexrj, 2017-04-13:
@lordofhyphens, there's little to factor out because Plater transforms whole ModelObjects, while ObjectPartsPanel would transform single ModelVolumes, so different API calls. I'd just write new code, not much is needed.

lordofhyphens, 2017-04-14:
@foreachthing there you go.

lordofhyphens, 2017-04-14:
@foreachthing http://dl.slic3r.org/dev/win/branches/slic3r-resize-mesh.2017.04.14.938.e202f5b.zip If you want to work with it a bit.

foreachthing, 2017-04-15:
@lordofhyphens
Thank you! I was like: "nah-damn, I can't build that branch!" and then you upload that - thanks!! :-)
So far it seams to work pretty good!!

But for some reason, I got some (Chinese?!) characters on the buttons.
[image]

lordofhyphens, 2017-04-15:
I will see what is going on, I think it is related to file encoding

lordofhyphens, 2017-04-15:
Should be fixed now.

PR #3879 (2017-04-15): Resize & Rotate modifier mesh
Can now resize and rotate modifier meshes. Accessed through the right-click menu (similar to plater).

Fixes #3875.
closed 2017-04-15 by lordofhyphens