slic3r/Slic3r #3876: v1.3.0-dev incompatible with Repetier Host Mac

RalphSutherland, 2017-04-11:
### Version
1.3.0-dev-5cc3d34.dmg

### Operating system type + version
MacOS 10.12.4

### Behavior
Interface with Repetier Host Mac v1.0.2 (102) is broken due to changed CLI flags

'Configure Slic3r' button won't launch Slic3r, and log shows the following errors followed by the CLI flags help.

  6:00:43 pm: <Slic3r> Unknown option: no-plater
  6:00:43 pm: <Slic3r> Unknown option: gui-mode
  6:00:43 pm: <Slic3r> Slic3r 1.3.0-dev is a STL-to-GCODE translator for RepRap 3D printers
  6:00:43 pm: <Slic3r> written by Alessandro Ranellucci <aar@cpan.org> - http://slic3r.org/
  6:00:43 pm: <Slic3r> Usage: slic3r.pl [ OPTIONS ] [ file.stl ] [ file2.stl ] ...
  6:00:43 pm: <Slic3r> --help              Output this usage screen and exit
.... 

Actual calls to *slice* do work as normal, in fact nicer than 1.29 :)

Workaround for now is to keep Slic3r open for editing configurations manually, and only use the slice button in Repetier Host app.  (side issue One reason to use the Repetier Host app, is  has a nicer perspective preview that can switch between iso and perspective with a click so my brain doesn't hurt , hint hint!  RH + Slic3r is a great combo)

* Steps needed to reproduce the problem

Always - just try the 'configure Slic3r' button.

* _If this is a command-line slicing issue, include the options used_

--no-plater and --gui-mode

The host software appears to have the flags hardwired into the object code and can't be modified or hacked without corrupting the OS signing

* _Expected Results_

Launch Slic3r without the plater IU allowing slicer settings to be modified and saved.

* _Actual Results_

Nothing except the log

_Is this a new feature request?_

No, the question is why the flags were changed/deleted?  Is it  now Repetier's problem or can Slic3r retain the --no-plater and --gui-mode flags for compatibility??  End users cannot fix this.

lordofhyphens, 2017-04-11:
We generally don't try to accommodate closed source programs, which would
include RepetierHost (yes, the host is closed source).

I suspect the options went away when @alexrj updated the GUI.

I have never understood why people want perspective camera mode in their
CAD toolchain, but it is easy to do and already has a issue in the tracker.
I will probably just add the option to get it out of the way and remove
another reason to bother with RH (slic3r has host functionality now).

lordofhyphens, 2017-04-11:
Wow, perspective is *useless* for doing anything on the plater.

If you want to mess around with 3DScene.pm (reference the OpenGL docs for more information), I have a hacked-together perspective view:
https://github.com/alexrj/Slic3r/tree/perpsective

lordofhyphens, 2017-04-11:
I was right, @alexrj removed the option when he did gui3
@xoan finished the job with https://github.com/alexrj/Slic3r/pull/3812 (because the option did nothing at that point).

lordofhyphens, 2017-04-11:
The original option for --no-plater and --gui-mode stems from #604 (note that RepetierHost was open source at this point). Up to him if he wants to put those options back in or not (if he doesn't this will be closed with wontfix tag).

alexrj, 2017-04-11:
We should at least accept and ignore those options, in order to launch the GUI.
`--gui-mode` was used to get the expert mode, which is now the only behavior, so not needed anymore.
`--no-plater` is more complicated to get back. Doable, though. While supporting closed source software is not a priority, retrocompatibility makes sense. I'll think about it.

alexrj, 2017-04-11:
Done, those options are now accepted and ignored so that GUI launches.

RalphSutherland, 2017-04-12:
Great! A fair and reasonable choice, thanks!  BTW  I've forked RHM v0.56, the last open version, and am going to look at what if anything of it can be evolved to an updated open version.    

PS I use isometric when in measurement/comparing mode and lining things up, but perspective when understanding the overalls of a complex shape and slicer paths, it is just easier for the my brain to process, pure personal preference.   

PPS The hosting for Slic3r is a great start, but I don't propose getting all heated about it, RHMac is just a good host and controller.   If I get deeper into Slic3r I may contribute, but Objective C/Cocoa  is more my speed and Perl is not so familiar :)   Cheers!

lordofhyphens, 2017-04-12:
We're slowly moving parts of Slic3r away from Perl ;)

lordofhyphens, 2017-04-12:
@repetier should probably at least know about this so they can deal with it if they want to.

closed 2017-04-12 by lordofhyphens

repetier, 2017-04-12:
I have just finished the repair for windows. One key point is that now you need to reference slic3r-console for slicing, otherwise the window comes up instead. And of course no plater option and --gui instead. I will rebundle a new mac host with the corrections once the stable slic3r 1.3 is out, which would be in april if you can hold the date announced.