slic3r/Slic3r #3878: Error when loading STL - Can't call method "objects_count" on an undefined value

bonzai95, 2017-04-13:
### Version
Slic3r Prusa Edition
Version 1.34.1-prusa3d-win64

### Operating system type + version
Windows 10 Version 1607  OS Build 14393.1066

### Behavior
Slic3r throws an error when trying to open this model http://www.thingiverse.com/thing:1533820 

[image]

#### STL/Config (.ZIP) where problem occurs
[dice_bot.zip](https://github.com/alexrj/Slic3r/files/919898/dice_bot.zip)

lordofhyphens, 2017-04-13:
Will reopen if you reproduce on the topmost build from dl.slic3r.org/dev

closed 2017-04-13 by lordofhyphens

bonzai95, 2017-04-13:
Looks like it is resolved in the top dev build

lordofhyphens, 2017-04-13:
Then you should report it to the prusa3d fork, it was either fixed here and
not there or it was introduced by their fork. Something they should know
about though.

In the future, please provide a reproduction test case with our code, not
the prusa3d fork, for issues reported here.