slic3r/Slic3r #3880: Display scaling in Windows not drawing GUI in high resolution

ade, 2017-04-14:
### Version
Windows: slic3r-master.2017.04.13.937.b78ccc6.zip

### Operating system type + version
Windows 7 x64
Using a 4k monitor, and 150% scaling in Control Panel\Appearance and Personalization\Display

### Behavior
The Slic3r GUI is upscaled and looks blurry, similar to the issue on retina/5k macs. Everything still works properly aside from that.

_Is this a new feature request?_
No

[image]

alexrj, 2017-04-14:
[wxWidgets 3.1.0 release notes](https://www.wxwidgets.org/news/2016/02/wxwidgets-3.1.0-released/) mention:

```
* Better support for high DPI displays, especially under Windows.
```

But I'm afraid our Windows build server already uses 3.1.0 (@lordofhyphens?)...

lordofhyphens, 2017-04-14:
Afaik it does use 3.1.0. An easy way to confirm is to look at the dlls.

lordofhyphens, 2017-04-14:
@ade  can you copy all of the .dll files from `local-lib\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib` (starting from the extracted archive) into the root Slic3r folder and see if that has any effect?

ade, 2017-04-14:
Tried it, that didn't make any difference.

fcannizzaro, 2017-08-06:
@ade I solved blurry text in Slic3r setting DPI scaling to "**Application**"

[image]

ade, 2017-08-09:
@fcannizzaro Thank you, but that option does not seem to be available in Windows 7.