slic3r/Slic3r #3882: perl524.dll missing from your computer

arkbox, 2017-04-15:
### Version
_Version of Slic3r used goes here_1.3

_Use `About->About Slic3r` for release versions_2017.04.13

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_win10 home

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_slic3r-master.2017.04.13.937.b78ccc6.zip

lordofhyphens, 2017-04-15:
http://dl.slic3r.org/dev/win/slic3r-master.2017.04.13.937.b78ccc6.zip has the file right there, so I don't know what's going on.

arkbox, 2017-04-15:
I have previous version 1.29 from repetitor host instulation. In slic3 folder I have 2 exe files and 4 folders and in cpfworkrt folder I have perl514.dll.  
In last 1.30 version folder I have 4 exe files and zero additional folders with dll. If I try to open exe I get perl524.dll missing from your computer message. perl514.dll from previous version don't help. Now it understand?

closed 2017-04-15 by arkbox

lordofhyphens, 2017-04-15:
If you're going to try to integrate with repetierhost (I don't use repetierhost at all and we don't support it), remove the old slic3r folder and extract the new version into a new slic3r directory. 

You need *every* file/folder that is in the new slic3r folder for it to run. Things have changed.

arkbox, 2017-04-25:
Thanks, but you steel don't develop supports with common level, I cant to use slic3r slicer.