slic3r/Slic3r #3888: Should "resolution" option be deprecated?

Tinchus2009, 2017-04-16:
### Version
latest dev version

### Operating system type + version
win10

I noticed this today, while accesing an option I used not all the time but somehow often: resolution
It called my atention that now it says "deprecated", nevertheless still it is working, but I hope no removed in the future?
Why should considere to keep this option: O lot of models comes with some sort os "watermark" signature on its walls. I have no problems at all with this or printing them, but many times these signatures are badly modelled and actually dont print and sometimes even cause troubles because they cerate some nasty blobs. 
My latest example would be this one (this one in particular doesnt cause printing troubles, but since is the one printed today Im going to use it as example). Useing  a normal 0.4 noozle, if you slice this simple model you will notice some weird path deviations. These deviations are actually not a bad slice but the software trying to slice this signature, wich is too small to be printed:

[image]

Slice:

[image]

These kind os small potential problems can easily be handled by the resolution option, just setting a value of 1, woud remove these kind of bad modelled details, with no other effect on the sliced model:

[image]

foreachthing, 2017-04-17:
"Resolution" could be replaced with "noise filter". I often print STLs from FEA simulations and that mesh is, most of the time, is rough, "bad", non-manifold and noisy ...

But to answer your question: I don't use "resolution", in other words, set to zero and I would not miss this feature.
I'm more of a change-my-model-'till-it-slices type of guy. I like the hit that my mesh is manifold, or not, but that's it.

alexrj, 2017-04-29:
Heh, you found an uncommon usage scenario.
It's deprecated because people set it to big values like 0.1mm and then complain "oh, Slic3r has bad quality"...

Tinchus2009, 2017-04-29:
Well, It is not uncommon to me. I use this option not Often, but for close to. I think the option is usefull, dont understand why people would use it and then complain if they dont know how to use it lol.
I havent seen this kind of option in other slicers, and as I showed, it is usefull with a lot of models out there
Just an opinion

bubnikv, 2017-04-30:
The issue with this setting is that produces a simplification, which is
incoherent between the layers. Something like a quadric edge collapse
method over a manifold mesh will produce a simplified mesh, leading to a
coherent simplification of the slices.

alexrj, 2017-04-30:
Exactly. This is why it's deprecated.