slic3r/Slic3r #3889: [Feature Request]Rectilinear Infill

powertide, 2017-04-17:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Tinchus2009, 2017-04-17:
Hi. You can already change the fill angle with the option "fill angle" . And why would you want change the infill angle in each layer for this pattern of infill? Doing as you want will only generate bridges in each layers since the infill in each new layer will be unsupported... Also , may be you are trying to get something like the cubic infill pattern maybe?

powertide, 2017-04-18:
Hi，thanks for your comment @Tinchus2009. According to what I know, the "fill angle" which slic3r already has in its settings might also be called exactly——infill start angle. It stands for the infill angle for the first infill layer. 
While the "**infill phase**" I said means—— **infill angle increment**, namely the angle  increment by each layer.
I don't quite understand the part where you said"Doing as you want will only generate bridges in each layers since the infill in each new layer will be unsupported".
Because the infill angle is already increasing by 90° by each layer, or I could say the infill phase is now 90° by default.
So the **question** here is: **Was the infill phase not chosen to open because the 90° infill phase has proved to always produce best printing?**
Because from what I can tell, sometimes people would like the infill angle and infill phase to be both adjustable, especially in other 3D-printing techs, such as SLM . And in my opinion it's best to open the setting for infill phase.

NateTG, 2017-04-19:
Suppose that the fill pattern is just parallel lines, and let's say that the lines on the bottom layer are all parallel to the X-axis.  Then if we rotate the lines by 90 degrees each layer (as it seems you're proposing) the fill pattern on the second layer would be lines that run parallel to the Y-axis so the fill on the second layer would only be supported by the fill on the bottom layer at grid intersections.  The rest of the second layer would be material without support (i.e. bridges.)

If that doesn't accurately describe what you're thinking of, maybe some kind of illustration would be helpful for explaining your idea.

FWIW: For any linear periodic pattern changing the phase corresponds to a linear displacement.  So "infill phase" is poor term for describing angle changes unless the pattern has circular symmetry.

powertide, 2017-04-20:
Thanks for your comment @NateTG . And yes, you decribed clearly and accurately about the infill angle change I proposed(BTW thanks for pointing out about the  "infill phase"). 
But note that the 90° infill angle change I mentioned is the setting by default in slic3r, not what I proposed.
You can see from below,this is a random layer:
[image]
And the next layer:(layer height=0.04):
[image]
And infill for the next after that looks exactly like the first one. So apparently, the infill angle increases by the default 90 degrees by each layer.

NateTG, 2017-04-21:
Huh..

I'm guessing that angle is set on line 446 of Slic3r/xs/src/libslic3r/Fill/FillRectilinear.cpp

https://github.com/alexrj/Slic3r/blob/bc35063a57788a23381b9e789289c27110fe07b7/xs/src/libslic3r/Fill/FillRectilinear.cpp

I think that changing "PI/2" on that line would give you a parallelogram grid instead of continuously twisting.  I'm also not sure how you'd go about setting up a way to change it in the UI.

I'm not sure how interface settings get passed to the fill routines,

lordofhyphens, 2017-04-21:
The UI modifies configuration options, which the code uses.

NateTG, 2017-04-21:
> The UI modifies configuration options, which the code uses.

I figured, but I don't really know how the plumbing (for lack of a better term) works.

lordofhyphens, 2017-04-21:
Refer to the config in the code, and so long as the option is registered
via printconfig and set up in the preset editor it will be available.

NateTG, 2017-04-22:
> Refer to the config in the code, and so long as the option is registered via printconfig and set up in the preset editor it will be available.

Things that are obvious to you, are not obvious to me.  I have no idea what the preset editor is.

Is there an example of a 'only for one thing' option that I can use as a reference?

lordofhyphens, 2017-04-22:
https://github.com/alexrj/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI
There's a Wiki entry with the files of note for adding a new CLI/GUI
option. That wiki page is linked from
https://github.com/alexrj/Slic3r/wiki/Documentation, which is in turn is
"Other Documentation" on the wiki sidebar.

If you think the wiki entry is lacking then feel free to fill it out when
you get a better feel for how the code works.

There's PrintRegion.cpp:90, which has a single option being referred to as
a member of the PrintRegionConfig class.

Layer.cpp:164 gets a reference to a PrintRegionConfig (which is defined in
PrintConfig.hpp:207). Every configuration option is a member variable for
that config type.

Because of the magic of inheritance, a FullPrintConfig "is-a"
PrintObjectConfig, PrintRegionConfig, PrintConfg, and a HostConfig.

PresetEditor.pm is new (and referred to on the wiki). Search for an
existing configuration option there and you can learn by example. It's just
the GUI for those items (which are seen in the GUI like the Infill %, etc).

The way I figured out how most things go together was by heavy use of the
"grep" tool to search for option names. Searching over the entire codebase
for the title of something brought me to the internal name, which got me
everywhere it was used. It's still generally how I explore code. If you're
on Windows then go looking for a tool to search text in entire directory
structures.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2017-04-29:
@powertide, so many things could be made configurable, but we need good reasons for each one, as Slic3r already has a huge amount of options. :)

Rectilinear infill is best used at 100% density, otherwise extrudates don't touch the previous layer except in the intersection points, which makes for weak objects regardless of the directions. At 100% density, alternating by 90° is supposed to provide maximum strength for geometric reasons.

Can you provide some discussion about the scenario you're thinking about? Also, you're free to modify the code and experiment yourself before proposing changes, which we might consider. If you want to do that, you just need to add an override for `_layer_angle()` in FillRectilinear.hpp. The base class is defined in Fill.hpp.

closed 2017-06-05 by powertide