slic3r/Slic3r #3890: Build System - Build libslic3r and then generate XS shared library linking to it?

lordofhyphens, 2017-04-18:
We already build libslic3r in ./src (the non-perl cli).

@alexrj is it possible to only generate the XS code and compile it (so perlglue and XS.c), and link against a static libslic3r (built with -fPIC) instead of relying on Module::Build to generate everything? It would be a useful step towards removing Module::Build entirely and would represent a large speed boost (as well as removal of duplicated build files)

Module::Build is quite opaque to me. so I'm not sure if this is viable.

lordofhyphens, 2017-04-22:
SWIG bindings are designed around this approach.

alexrj, 2017-04-29:
XS.c is generated dynamically by Module::Build and its dependencies, so not sure how far you'd go. We could compile the non-XS libslic3r separately and link to it, but it's a good amount of work and I don't think it's worth.
What's your goal? I don't think removing Module::Build is a goal, it does its job while we have the Perl/XS architecture. Let's work on the C++ port instead... :)

closed 2017-05-04 by lordofhyphens