slic3r/Slic3r #3891: f25ea9f4936f24399d5d657b7f4650db1529c8d6 broke perimeter pressure control

VanessaE, 2017-04-18:
### Version
git commit 3776f1afc64d5bcf8079b4e83e0376a79f91573e

### Operating system type + version
Debian sid

### Behavior

Because Slic3r now prints external perimeters at their configured speed even when slow-down is invoked for a given layer, this causes a sudden increase in speed at the start of an external perimeter, relative to the speed the inner perimeters and infill were just printed at.  This causes the start of the external perimeter to be starved of filament.

Pressure advance can compensate for this only in part, but if you set it high enough to fill-in the starved areas, it ruins pretty much everything else about the print that depends on proper pressure control.

#### STL/Config (.ZIP) where problem occurs

No configs or models necessary, just set all perimeter/infill/etc speeds to the same, set the cooling threshold "slow down" feature in the filament profile to a high time limit, say 45-60 seconds, and print a test cube.  Probably needs a bowden-fed printer to be noticeable.

Tinchus2009, 2017-04-18:
Have you tried changeing the acceleration for perimeters? that way you might achieve to reduce that sudden change of speed at the beginning of the perimeter. Im definitly not observing this problem in my printer, but as you said, this might be only affecting bowden xtruders and I have a direct one. In fact the change regarding this issue has improved the surfaces of my prints, some banding was happening there and this change made them desapear

VanessaE, 2017-04-18:
Acceleration isn't going to help here - in a simple test object, such as a cube or cylinder in the 2-3cm diameter range, two thirds of each line is starved.  The problem stems from the inner perimeters being printed at one speed, then the outer perimeter is printed perhaps 4 times faster, but without the requisite pressure boost to account for the change in speed.

Disable the slow-down feature and the starvation goes away (but then some other way to extend the layer time is needed to prevent overheating on small parts).

Tinchus2009, 2017-04-18:
I dont have a bowden extruder to test what I said, but acceleration shuld help there. In first place, you wouldnt expect much quality printing at 100 mm/s. Second, if the inner perimeter is printing at 50 mm/s and the outer at 100, setting an acceleration really low like 50should help sin ce the speed increase will give time for the pressure to keep up the accel... And again going back to 1, it looks like you are trying to get a solution for an unreal problem: the slow down option is necesary for cooling porpuses, if you know that you need cooling why setting a 100 mm/s speed ? I mean... there is a wall in front of you, if you dont stop you are going to crash, instead of that you send a letter to the car company complaining about a missing teletransportation device not installed in the car? (this is a joke, dont get ungry ). 
The pressure control in the outer perimter is one of the improvements that increased quality in the surface and it has been working really well, the situation you are presenting is somehow real but not a wise way to configure a printing. If the slow down option gets triggered that means the object is small, you dont need to use a value greater than 20 seconds, that is more than enough time to let pla cool and still you have a cooling fan. If it gets triggered with that value that means the object is small, so why do you want to use a 100 mm/s speed for such an object...

lordofhyphens, 2017-04-18:
I don't think it was this commit specifically. If it was, it just exposed something else that was going on.

VanessaE, 2017-04-18:
I only blame that commit because before, I had *everything* set to the same speeds and extrusion widths, to basically eliminate the pressure changes.

closed 2018-04-25 by VanessaE