slic3r/Slic3r #3892: Slic3r Automated development builds broken

kbm15, 2017-04-18:
All new files are empty in http://dl.slic3r.org/dev/win/

lordofhyphens, 2017-04-18:
I can't tell if this is Appveyor's fault or something is going weird on the server, the .filepart  extension means that it never finished uploading.

@alexrj any ideas?

lordofhyphens, 2017-04-18:
@alexrj We need that cron job to purge old builds, I suspect. I'll take a look at when I can sftp into the system next whether or not deleting old files helps.

lordofhyphens, 2017-04-18:
@alexrj Yes, we did run out of space on dl.slic3r.org

lordofhyphens, 2017-04-18:
@alexrj 
For purposes of cron:
````
for i in $(ls -trl /path/to/folder/*.zip | awk '{print $9}' | head -n$(expr $(ls /path/to/folder/*.zip | wc -l) - 10)); do
    rm /path/to/folder/$i
done
````
will delete all but the 10 newest files ending in zip in /path/to/folder

Wrapping that in a script:

````
#!/bin/bash

FOLDER=$1
PATTERN=$2
COUNT=$3

for i in $(ls -trl $FOLDER/$PATTERN | awk '{print $9}' | head -n$(expr $(ls $FOLDER/$PATTERN | wc -l) - $COUNT)); 
do 
    rm -f $i; 
done
````
Usage: `rmoldest.sh FOLDER 'PATTERN' COUNT`
Get cron to run that (probably weekly), and wrap the pattern in single quotes when you call it (so *.zip -> '*.zip')

For now I'd have cron execute it daily on each of the dev/ paths. and monthly on dev/*/branches

lordofhyphens, 2017-04-18:
in gist form: https://gist.github.com/lordofhyphens/0acb46e9557e38d2c002f821e11c8bea

closed 2017-04-22 by lordofhyphens