slic3r/Slic3r #3893: Honeycomb infill does not respect `Infill/perimeter overlap` setting

foreachthing, 2017-04-19:
### Version
slic3r-master.2017.04.19.958.3274b75

### Operating system type + version
Windows 7 Pro

### Behavior
* Print Settings
  * Infill
    * Fill pattern: **Honeycomb** or **3D Honeycomb**
  * Advanced
    * Infill/perimeter overlap set to anything or leave unchanged ( I set it to 85% to see the difference)

Now, in Layers-Tab you can clearly see, that the Honeycomb infills do not get as close to the perimeter as it should. It is working 1.2.9.41 (I didn't test any other versions).
**All other infill patterns work as expected.**
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
Any

foreachthing, 2017-09-07:
_Bump._
Still no response? Can anyone confirm this issue or is it just me?

bbrummer, 2018-01-21:
Appears ignored for me as well, build Slic3r-master.2017.12.21.1566.6dd4009.64bit.zip

:(