slic3r/Slic3r #3894: Using the prebuilt Perl dependency archive

RobWks, 2017-04-19:
### Version
1.2.9 

### Operating system type + version
Windows 10

### Behavior
I am attempting run Slic3r from git following these instructions (https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows). Everything seems to build without error (including the GUI) and the command line interface seems to work. The issue arises when trying to run the GUI, the Perl interpreter crashes and Slic3r becomes unresponsive. When running Slic3r with --debug, it shows that I am using wxWidgets 3.0.2 which I believe is the source of the issue. The instructions state to use the prebuilt dependency archive which includes wxWidgets 3.1.0.  I am quite new to Perl, how do I actually use the local-lib folder in the dependency archive in order to run Slic3r with wxWidgets 3.1.0?

Thanks!

-Rob

lordofhyphens, 2017-04-19:
The archive has a single directory in it, local-lib. Put it into your
Slic3r source directory, overwrite when asked.

lordofhyphens, 2017-04-19:
Do not run perl Build.pl --gui ever.

RobWks, 2017-04-19:
So I started from scratch and added local-lib to C:/Slic3r-1.2.9/. Built fine using `perl Build.pl`, but I get the following error when running `perl slic3r.pl --gui`. Does `Build.pl --gui` need to be run once before placing in local-lib or am I missing something?

``
Can't locate Wx.pm in @INC (you may need to install the Wx module) (@INC contain
s: C:/Slic3r-1.2.9/lib C:/strawberry-perl-5.24.1.1-32bit-portable/perl/site/lib/
MSWin32-x86-multi-thread-64int C:/strawberry-perl-5.24.1.1-32bit-portable/perl/s
ite/lib C:/strawberry-perl-5.24.1.1-32bit-portable/perl/vendor/lib C:/strawberry
-perl-5.24.1.1-32bit-portable/perl/lib .) at C:/Slic3r-1.2.9/lib/Slic3r/GUI/Abou
tDialog.pm line 6.
BEGIN failed--compilation aborted at C:/Slic3r-1.2.9/lib/Slic3r/GUI/AboutDialog.
pm line 6.
Compilation failed in require at C:/Slic3r-1.2.9/lib/Slic3r/GUI.pm line 8.
BEGIN failed--compilation aborted at C:/Slic3r-1.2.9/lib/Slic3r/GUI.pm line 8.
Compilation failed in require at (eval 83) line 1.
``

lordofhyphens, 2017-04-19:
Why are you using 32bit strawberry perl? Unless you have a good reason to,
use the slic3r-perl distribution as the wiki says.

The dependency archive and perl version *must* match or you will have a bad
time.

Additionally, is this actually 1.2.9 or 1.3.0-dev?

(If you don't need to modify the source code, use a build from
http://dl.slic3r.org/dev/win).

RobWks, 2017-04-19:
So I'm trying to set this up on a 32bit Windows system and I get an error from windows that the `This installation package is not supported by this processor type. Contact your product vendor.` when using the slic3r-perl distribution, so I was looking for an alternative. I was using the dependencies linked here: https://bintray.com/lordofhyphens/Slic3r/Slic3r_Dependencies/Dependencies-1.3.0-dev
Which I believe should work with Perl 5.24.0 (32bit)?

I just noticed that this was actually for 1.3.0-dev though, I could repeat the process for 1.3.0-dev? I have modified the source code with a custom infill which works with the command line but I am looking at getting the GUI working for another user.

RobWks, 2017-04-20:
I ended up giving up installing it on that system and went with a 64bit installation on another computer. Everything seems to work great with the latest development version of Slic3r. Thanks for your help though lordofhyphens!

closed 2017-04-20 by RobWks