slic3r/Slic3r #3895: Setting "Overhang threshold" to any nonzero value creates undesired interface pattern artifacts

Gatulibu, 2017-04-20:
_Expected Result_:
The topmost interface pattern is rectilinear:

[image]
_Version used 1.2.9 stable (W7-x64), all settings default + support enabled._

_Actual Result_ with "Overhang threshold" set to 25:
The topmost interface layer seems to have some hexagonal lines/artifacts at the outer part of the support:

[image]
_Version used 1.2.9 stable (W7-x64), all settings default + support enabled + Overhang threshold set to 25°._

The new % setting introduced in the dev versions seems to create the undesired pattern with every value I tried.
[image]
_Version used dev (2017.04.19.958.3274b75 W7-x64), all settings default + support enabled._

Test STL:
supporttest.stl
[supporttest.zip](https://github.com/alexrj/Slic3r/files/939911/supporttest.zip)

lordofhyphens, 2017-04-20:
As a FYI: 200% is equivalent to the old value of 0 (or auto).

foreachthing, 2017-04-20:
> The topmost interface layer seems to have some hexagonal lines/artifacts at the outer part of the support

I agree ... to a point: I, too, think there is _too much support_ (in X and Y) and no way to control that. I don't think of those jagged lines as artifacts as much as unnecessary. Especially outside the part area to support - where it's only wasted plastic.
The user should be able to control the offset (X and Y) of the support material (as one can in Cura).

lordofhyphens, 2017-04-20:
There is a PR for that, actually. It is pretty hacky though.