slic3r/Slic3r #3898: crash while opening an stl file

sapir, 2017-04-21:
### Version
1.2.9-947-gef8695d

### Operating system type + version
Ubuntu 16.10, 32-bit

### Behavior
Slic3r crashes while opening the attached stl file

console output is as follows: (the messages up to but not including the assertion error are printed while opening other files, too, so maybe irrelevant)
```
16:55:13: Warning: Mismatch between the program and library build versions detected.
The library used 3.0 (wchar_t,compiler with C++ ABI 1009,wx containers,compatible with 2.8),
and your program used 3.0 (wchar_t,compiler with C++ ABI 1010,wx containers,compatible with 2.8).

(slic3r.pl:10222): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",

(slic3r.pl:10222): Gtk-WARNING **: Unable to locate theme engine in module_path: "adwaita",
perl: malloc.c:3757: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.
Aborted (core dumped)
```

#### STL/Config (.ZIP) where problem occurs
[crashing-stl.zip](https://github.com/alexrj/Slic3r/files/947265/crashing-stl.zip)

sapir, 2017-04-21:
forgot to mention: stl was output by the blender 2.74 stl plugin. importing then exporting with meshlab 1.3.2 produces an stl file that doesn't crash.

lordofhyphens, 2017-04-21:
This is a local build problem, you used a different c++ compiler to build
Slic3r than what was used to build wxperl and perl.

It is not a bug in Slic3r.

lordofhyphens, 2017-04-21:
Well, the warning isn't a slic3r bug anyway.

lordofhyphens, 2017-04-22:
Garbage in; garbage out.

closed 2017-04-22 by lordofhyphens

sapir, 2017-04-22:
you mean the crash is also due to the compiler mismatch?

lordofhyphens, 2017-04-22:
Crash is due to the model being garbage. The compiler mismatch is
incidental.

sapir, 2017-05-01:
But surely Slic3r should give some error message instead of crashing, even
if the model is garbage and it can't fix it?