slic3r/Slic3r #3900: Resize part under settings doesn't affect print

kefir-, 2017-04-22:
### Version
1.3.0-dev f24fc2c6757a819ceaf5344bb6ecd5479b0eacb1

### Operating system type + version
Ubuntu Linux

### Behavior
1. Add a part to the print bed.
2. Right-click part, choose Settings...
3. Right-click part again in the pop-up window, choose "Scale to size -> Along Z-axis"
4. Type in a value and click ok, and see that the preview in this pop-up has changed
5. Click OK to close pop-up, and now the main part is unchanged

I would expect the main part to be changed as well. If I do the same directly, without opening the Settings window first, it works as expected.

Screenshot shows difference in the two windows, here the Settings window was reopened after clicking OK, and Slic3r remembers the size of the part there.

[image]

_Is this a new feature request?_
NO

#### STL/Config (.ZIP) where problem occurs
Not required

lordofhyphens, 2017-04-22:
Move the part or click arrange, see if it updates

kefir-, 2017-04-23:
If I click "Arrange" the part immediately updates.

NateTG, 2017-04-23:
Could this be a dup of this issue: https://github.com/alexrj/Slic3r/issues/3798

alexrj, 2017-04-29:
Fixed. Thank you!

closed 2017-04-29 by alexrj