slic3r/Slic3r #3901: Changing parameters changes view to 3D-tab

kefir-, 2017-04-22:
### Version
1.3.0-dev f24fc2c6757a819ceaf5344bb6ecd5479b0eacb1

### Operating system type + version
Ubuntu 16.04

### Behavior
With the new interface, I'm changing some quick some quick-access parameters in the main window. Every time I change a setting, eg changing from 3 to 2 perimeters, the view tab changes from my preferred "Preview" to "3D". This seems to happen every time.

I would prefer for it to remain on view tab I have chosen, because this happens all the time and I want to verify the Preview to see that the change had the desired effect or if I need to change more.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
Not required.

gege2b, 2017-04-22:
if I remember, it's a wanted behaviour if background processing is disabled (but I agree with you, it's a bit annoying)

switching to preview tab fire the "slice now" function, it's a way to avoid adding a button in the GUI and still have a "slice now like" function

If you want to keep preview active when changing parameters, you need to enable the background processing

lordofhyphens, 2017-04-22:
Turn on background processing. This is expected behavior with background
processing off.

closed 2017-04-22 by lordofhyphens