slic3r/Slic3r #3902: idea of adding new feature for real colour mixing

sebastiangonzalezv, 2017-04-22:
Marlin firmware actually has a command 
M165: Set multiple mix weights
    A[factor] Mix factor for extruder stepper 1 
    B[factor] Mix factor for extruder stepper 2 
    C[factor] Mix factor for extruder stepper 3 
    D[factor] Mix factor for extruder stepper 4 
    H[factor] Mix factor for extruder stepper 5 
    I[factor] Mix factor for extruder stepper 6 

were the idea is use A for cyan, B magenta, C yellow, D black, H white, I support material.
the idea is when there is a mixing nozzle like
[image]
or 
[image]

can create a color mixing like or even more
[image]

here is more documentation of 3d color mixing

[image]

so the point is use the same actualg code generation but use instead t1 t2 t3...  to  change extrude use
use m165 to define a color
http://www.reprap.org/mediawiki/images/0/05/RepRapColourMixingReport-jmc.pdf
http://www.reprap.org/mediawiki/images/a/a5/Pia-taubert-material-mixing-report.pdf

lordofhyphens, 2017-04-22:
Pretty sure this is a dupe. Will go looking later.

Kadah, 2018-12-29:
#3456