slic3r/Slic3r #3903: Compile perl on OSX 10.12 failed

ankang1943, 2017-04-23:
`Failed 1 test out of 2454, 99.96% okay.
	../cpan/libnet/t/hostname.t
### Since not all tests were successful, you may want to run some of
### them individually and examine any diagnostic messages they produce.
### See the INSTALL document's section on "make test".
### You have a good chance to get more information by running
###   ./perl harness
### in the 't' directory since most (>=80%) of the tests succeeded.
### You may have to set your dynamic library search path,
### DYLD_LIBRARY_PATH, to point to the build directory:
###   setenv DYLD_LIBRARY_PATH `pwd`; cd t; ./perl harness
###   DYLD_LIBRARY_PATH=`pwd`; export DYLD_LIBRARY_PATH; cd t; ./perl harness
###   export DYLD_LIBRARY_PATH=`pwd`; cd t; ./perl harness
### for csh-style shells, like tcsh; or for traditional/modern
### Bourne-style shells, like bash, ksh, and zsh, respectively.
Elapsed: 883 sec
u=7.89  s=4.21  cu=480.71  cs=57.04  scripts=2454  tests=1241630
make: *** [test] Error 1`

lordofhyphens, 2017-04-23:
Not a Slic3r bug.

alexrj, 2017-05-10:
Yeah, not a Slic3r bug, and also very hard to help you if you don't post the command(s) you issued. @ankang1943, are you going to add any details about this?

lordofhyphens, 2017-05-21:
@kannng Anything else to add? I'm going to close this for now and if you have more information I can reopen.

closed 2017-05-21 by lordofhyphens

kannng, 2017-05-22:
@lordofhyphens @alexrj I'm so sorry to recover you so late I have solved this problem, but the solution is very stupid, because some of the last resort I reset my system, and then the problem does not exist. There was another problem when building gui, but got the answer in another Issue. Thank you very much for the concern of this issue, do not have to worry about this.