slic3r/Slic3r #3904: Slic3r sometimes forgets to lift Z on moves?

kefir-, 2017-04-23:
### Version
1.3.0-dev f24fc2c6757a819ceaf5344bb6ecd5479b0eacb1

### Operating system type + version
Ubuntu 16.04

### Behavior
I'm printing 5 hollow cubes arranged along the X axis. This is not spiral vase, the objects have 2 perimeters and a 2 layer base.

[image]

It appears that Slic3r forgets my Z lift option sometimes? But not on all layers. For example, I'm printing with 0.2mm layer heights, and would expect Z lift on every layer. Here's some of the results when counting all the Z restore moves in the gcode file:

```$ grep "restore layer Z" calibrationcube_20170423-141252_1.3.0-dev.gcode | sort -n | uniq -c | sort -rn | less
[snip]
      5 G1 Z9.850 F6000.000 ; restore layer Z
      5 G1 Z9.650 F6000.000 ; restore layer Z
      5 G1 Z9.450 F6000.000 ; restore layer Z
      5 G1 Z9.250 F6000.000 ; restore layer Z
      5 G1 Z9.050 F6000.000 ; restore layer Z
      5 G1 Z8.850 F6000.000 ; restore layer Z
      5 G1 Z8.650 F6000.000 ; restore layer Z
      5 G1 Z8.450 F6000.000 ; restore layer Z
      5 G1 Z8.250 F6000.000 ; restore layer Z
      5 G1 Z8.050 F6000.000 ; restore layer Z
      5 G1 Z7.650 F6000.000 ; restore layer Z
      5 G1 Z7.250 F6000.000 ; restore layer Z
      5 G1 Z6.850 F6000.000 ; restore layer Z
      5 G1 Z6.450 F6000.000 ; restore layer Z
      5 G1 Z6.050 F6000.000 ; restore layer Z
      5 G1 Z5.650 F6000.000 ; restore layer Z
      5 G1 Z5.250 F6000.000 ; restore layer Z
[snip]
```
So the "restore layer Z" moves are missing from layer 5.450, 5.850, 6.250, 6.650, 7.050, 7.450 and 7.850 in this example. But it's not clear exactly why these layers are missing and the others are there.

#### STL/Config (.ZIP) where problem occurs
[calibrationcube.zip](https://github.com/alexrj/Slic3r/files/949683/calibrationcube.zip)

lordofhyphens, 2017-04-23:
Z lift will only occur if a retraction occurs (IIRC), and for short moves
there wouldn't necessarily be a retraction.

Is that behavior consistent with your observations?

lordofhyphens, 2017-04-23:
Also, do those missing restore moves correlate to the layer changes?

kefir-, 2017-04-23:
No, I don't think the missing lifts are consistent with short moves. Here's an example:

```
G1 X90.188 Y134.822 E6.71643 ; perimeter
G1 X90.188 Y124.542 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X90.410 Y124.219 F6000.000 ; move inwards before travel
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X125.472 Y134.058 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X116.952 Y134.058 E0.76531 ; perimeter
G1 X116.952 Y115.936 E1.32964 ; perimeter
```

kefir-, 2017-04-23:
Here's another one without Z lift, this move is approx 45mm along X axis and 12mm along Y:
```
G1 X58.980 Y115.235 E6.67954 ; perimeter
G1 X58.980 Y127.085 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X58.753 Y127.404 F6000.000 ; move inwards before travel
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X103.534 Y115.936 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X109.074 Y115.936 E0.67253 ; perimeter
G1 X109.074 Y134.058 E1.23685 ; perimeter
```

and here's one that does Z lift, which is also just over 40mm along the X axis and about 8mm along Y:
```
G1 X83.838 Y115.172 E6.62775 ; perimeter
G1 X83.838 Y129.227 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X83.604 Y129.540 F6000.000 ; move inwards before travel
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z2.650 F6000.000 ; lift Z
G1 X40.094 Y121.345 F6000.000 ; move to first perimeter point
G1 Z2.450 F6000.000 ; restore layer Z
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X40.094 Y115.999 E0.66648 ; perimeter
G1 X58.216 Y115.999 E1.23081 ; perimeter
G1 X58.216 Y134.121 E1.79513 ; perimeter
```

kefir-, 2017-04-23:
Here's some code from layer 7.450, that's definitely doing the wrong thing, with my comments:

Move to layer 7.450, and print the first box at X range [64 - 83]. Note, by the way, that this lift is commented as "lift Z", and not "move to next layer (XXX)" which is the comment I would have expected.

```
G92 E0 ; reset extrusion distance
G1 Z7.450 F6000.000 ; lift Z
G1 X64.952 Y118.980 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X64.952 Y115.936 E0.59478 ; perimeter
G1 X83.074 Y115.936 E1.15910 ; perimeter
G1 X83.074 Y134.058 E1.72343 ; perimeter
G1 X64.952 Y134.058 E2.28775 ; perimeter
G1 X64.952 Y119.032 E2.75567 ; perimeter
M204 S3000 ; adjust acceleration
G1 X64.545 Y119.032 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F3000
G1 X64.545 Y115.529 E2.86476 ; perimeter
G1 X83.481 Y115.529 E3.45444 ; perimeter
G1 X83.481 Y134.465 E4.04412 ; perimeter
G1 X64.545 Y134.465 E4.63379 ; perimeter
G1 X64.545 Y119.085 E5.11275 ; perimeter
M204 S3000 ; adjust acceleration
G1 X64.188 Y119.085 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F1800
G1 X64.188 Y115.172 E5.20466 ; perimeter
G1 X83.838 Y115.172 E5.66626 ; perimeter
G1 X83.838 Y134.822 E6.12786 ; perimeter
G1 X64.188 Y134.822 E6.58946 ; perimeter
G1 X64.188 Y119.137 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X64.426 Y118.828 F6000.000 ; move inwards before travel
```

Next, retract and move to the box at X range [39-58]:
```
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X49.668 Y115.999 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X58.216 Y115.999 E0.76620 ; perimeter
G1 X58.216 Y134.121 E1.33052 ; perimeter
G1 X40.094 Y134.121 E1.89485 ; perimeter
G1 X40.094 Y115.999 E2.45917 ; perimeter
G1 X49.615 Y115.999 E2.75567 ; perimeter
M204 S3000 ; adjust acceleration
G1 X49.615 Y115.592 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F3000
G1 X58.623 Y115.592 E3.03618 ; perimeter
G1 X58.623 Y134.528 E3.62585 ; perimeter
G1 X39.687 Y134.528 E4.21553 ; perimeter
G1 X39.687 Y115.592 E4.80521 ; perimeter
G1 X49.563 Y115.592 E5.11275 ; perimeter
M204 S3000 ; adjust acceleration
G1 X49.563 Y115.235 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F1800
G1 X58.980 Y115.235 E5.33397 ; perimeter
G1 X58.980 Y134.885 E5.79557 ; perimeter
G1 X39.330 Y134.885 E6.25717 ; perimeter
G1 X39.330 Y115.235 E6.71877 ; perimeter
G1 X49.510 Y115.235 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X49.834 Y115.456 F6000.000 ; move inwards before travel
```

Now, retract and move to print box at X range [90-109]. THis is a fairly long move, and moves over the first box without lifting:
```
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X105.269 Y134.058 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X90.952 Y134.058 E0.94583 ; perimeter
G1 X90.952 Y115.936 E1.51016 ; perimeter
G1 X109.074 Y115.936 E2.07449 ; perimeter
G1 X109.074 Y134.058 E2.63881 ; perimeter
G1 X105.321 Y134.058 E2.75567 ; perimeter
M204 S3000 ; adjust acceleration
G1 X105.321 Y134.465 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F3000
G1 X90.545 Y134.465 E3.21581 ; perimeter
G1 X90.545 Y115.529 E3.80549 ; perimeter
G1 X109.481 Y115.529 E4.39517 ; perimeter
G1 X109.481 Y134.465 E4.98485 ; perimeter
G1 X105.374 Y134.465 E5.11275 ; perimeter
M204 S3000 ; adjust acceleration
G1 X105.374 Y134.822 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
G1 F1800
G1 X90.188 Y134.822 E5.46948 ; perimeter
G1 X90.188 Y115.172 E5.93108 ; perimeter
G1 X109.838 Y115.172 E6.39268 ; perimeter
G1 X109.838 Y134.822 E6.85428 ; perimeter
G1 X105.426 Y134.822 E6.95792 ; perimeter
M204 S3000 ; adjust acceleration
G1 X105.085 Y134.625 F6000.000 ; move inwards before travel
```

Not much more to say here, retract and move to next box, etc....
```
G1 E6.45792 F1200.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X135.074 Y130.285 F6000.000 ; move to first perimeter point
G1 E0.50000 F1200.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F3000
G1 X135.074 Y134.058 E0.61749 ; perimeter
G1 X116.952 Y134.058 E1.18181 ; perimeter
G1 X116.952 Y115.936 E1.74614 ; perimeter
G1 X135.074 Y115.936 E2.31046 ; perimeter
G1 X135.074 Y130.233 E2.75567 ; perimeter
M204 S3000 ; adjust acceleration
G1 X135.481 Y130.233 F6000.000 ; move to first perimeter point
M204 S2000 ; adjust acceleration
[snip]
```

alexrj, 2017-04-27:
@kefir-, you have retract_lift == layer_height and also retract_layer_change enabled. In this case, the "move to next layer" lines are skipped because the lift performed before layer change already brought Z to the next layer height.
So, the restores you see are in-layer moves (5 per layer, which makes sense since you have 5 objects).
So this is not a bug. :)

alexrj, 2017-04-27:
Uhm wait. I'm digging further.

kefir-, 2017-04-27:
Thanks. :) This definitely happens on moves within a layer.

alexrj, 2017-04-29:
Okay, disregard my previous commit. There was actually a bug. I worked on stronger regression testing and this now works properly. Thank you :)

closed 2017-04-29 by alexrj

kefir-, 2017-04-29:
Thanks, I'll try it out!