slic3r/Slic3r #3905: Too long menu for quick access override settings.

kefir-, 2017-04-23:
### Version
1.3.0-dev f24fc2c6757a819ceaf5344bb6ecd5479b0eacb1

### Operating system type + version
Ubuntu 16.04

### Behavior
When overriding settings for the current print with the quick access settings, I click the small green "plus" to access the menu. The menu that pops up is so long, and I have to push the pointer up and down to scroll while search for and accessing the settings that I want. It's not many clicks, but there's a lot of pointer manipulation and accurate aiming involved. Repetitive tasks take a long time, and this design doesn't lend itself well to keyboard shortcuts.

[image]

I'm not a GUI guy, so I can't claim to know what is a better design. But I think using menus and submenus would be an improvement, so that the first menu is just 8 entries:

(+)
-> Advanced
-> Extruders
-> Extrusion Width
etc..

And to access the settings under Extruders, you'd have to click the plus, and then click/hover over Extruders, and then find these settings.

Also, the little green plus is small and arguably requires accurate aiming to click on. It's also less accessible when it disappears below the fold and I need to scroll down to find it. In this screenshot, I need to scroll down to find it.

[image]

alexrj, 2017-04-27:
@kefir-, I find much faster to click on the cog icon next to the profile menu and change settings from there...

kefir-, 2017-04-27:
I agree, but doesn't that defeat the purpose of overriding one of the base settings for the current print vs actually changing the settings? I thought that was the idea of the settings configured next to the preview.

lordofhyphens, 2017-04-27:
The profile would still be dirty, do nothing would be permanent.

alexrj, 2017-04-29:
Okay, we now have nice submenus:

[image]

closed 2017-04-29 by alexrj