slic3r/Slic3r #3906: OctoPrint integration broken in latest version

th0mpy, 2017-04-23:
### Version
slic3r-1.3.0-dev-f24fc2c

### Operating system type + version
OSX 10.12.4

### Behavior
The problem comes up with an attempt to send code to OctoPrint. I've confirmed both the IP and API key. I've attempted to use both an IP and host name, the same behavior occurs. The error I receive says:
_Error while connecting to the OctoPrint server: 500 LWP::Protocol::request() needs to be overridden in subclasses_

[image]

#### STL/Config (.ZIP) where problem occurs
[config and stl.zip](https://github.com/alexrj/Slic3r/files/949976/config.and.stl.zip)

alexrj, 2017-04-29:
Okay, this is a packaging issue. Some module is not being included.

alexrj, 2017-04-29:
Fixed. I only tested the new OSX package, but I applied the same fix to the other platforms too. Thank you for reporting the issue!

closed 2017-04-29 by alexrj