slic3r/Slic3r #3910: Weird moves in generated gcode with "Avoid crossing perimeters"

dl6lr, 2017-04-26:
### Version
slic3r-master.2017.04.21.968.f24fc2c.zip

### Operating system type + version
Windows 10 Pro Version 1607 Build 14393.1066 #s_

### Behavior
* Slicing the STL file (this is downloaded from Thingiverse 1347886 and repaired by netfabb) with "Avoid crossing perimeters" creates a gcode with a lot of unneccessary long tool moves without extrusion. Furthermore, it crosses parts just extruded at the same layer.
* To reproduce: Enable "Avoid crossing perimeters"
* Expected: Tool paths stay outside of objects, no unnecessary long moves.
* Actual: Weird moves around objects and all over the print area. First layer, to identify the parts I count rows from left to right as x1..x6 and from bottom to top as y1..y6:
[image]

Detail at the beginning, it moves from x2y6 to x2y5 around a not yet existing part x1y5, where the direct move would be shorter and faster. Additionally the generated gcode is awfully slow if printed via serial connection due to the move constructed from hundreds of small moves (my generated file has the move from line 922 to 1405):
[image]

Detail mid print of first layer, where the extruder moves from x6y1 to x1y1crosseing x5y1 and x2y1:
[image]

The same behaviour is found on the other layers as well.
If any information is missing, I will try to provide it.

#### STL/Config (.ZIP) where problem occurs
[weirdmoves.zip](https://github.com/alexrj/Slic3r/files/959646/weirdmoves.zip)

alexrj, 2017-05-16:
In the recent 893cab752362f58f49024e354a40f96dc4eff43e commit I restored travel path simplification which reduces the number of tiny segments you reported.

I'll check your test case, especially about those crossings, however I need to point out that the logic behind the "Avoid crossing perimeters" option is supposed to consider also the previous layer(s) and not just the current one. It's meant for leaky extruders.

dl6lr, 2017-05-19:
Hi alexjr,

I can confirm the latest dev build slic3r-master.2017.05.19.1075.aa3153f simplifies the tool path, the generated GCode is shorter. 
I have another STL that creates travels across the build area and crosses already printed perimeters of the same layer without need, See the peaks to the left print area, this is the first layer but the same is repeated on the others:

[image]

The outer perimeters and the circles on the left are printed just before.

Find the STL and the config here.
[weird2.zip](https://github.com/alexrj/Slic3r/files/1015509/weird2.zip)

mamoit, 2017-12-07:
I'm running the latest master commit (8ca6171).
I just tried slicing a variant of [this model](https://www.thingiverse.com/thing:255924) just with less scales, and after the flat part, the extruder goes in and out of the model area just to travel a couple of mm zigzaging between the scales.
If I turn off Avoid Crossing Perimeters, everything is fine and dandy.

smartynov, 2020-01-30:
I can confirm this is still the issue with the latest dev jenkins build (`slic3r-1.3.1-dev-c2aa7e7b-osx.dmg`). I think this feature is not a first priority, but it might help is some prints, so improving it should be somewhere on the backlog.

Typho81, 2021-03-28:
Confirmed with the version Slic3r-master.2021.03.15.2481.18ddd3a3.64bit-PR 

[image]
Avoid crossing perimeters enabled

[image]
Avoid crossing perimeters disabled