slic3r/Slic3r #3913: FFC Pro Sailfish Dual Material gcode incorrect 1.29

mron, 2017-04-27:
### Version
1.29

### Operating system type + version
osx and linux

### Behavior

Dual material print, Loaded object 1, in settings loaded object 2 with object 1
in "Printer Settings" selected Sailfish( Makerbot) as gcode flavor
I examined gcode for tool changes or T0 or T1 codes, none found beyond first T0.

#### STL/Config (.ZIP) where problem occurs
simple cubes.

lordofhyphens, 2017-04-27:
Use 1.3.0 dev, load the plate and export the whole thing as AMF and upload with a copy of your config.

lordofhyphens, 2017-04-27:
dl.slic3r.org/dev

mron, 2017-04-28:
I included a simple amf, the gcode output with sailfish and raprap flavors, and the config files for both sailfish and reprap

[Archive.zip](https://github.com/alexrj/Slic3r/files/964965/Archive.zip)

mron, 2017-04-29:
I think the problem is not with slic3r, but with gpx, or specifically, my inability to get gpx to convert gcode to sailfish compatible x3g. There is a M108 in the gcode, which i think is sailfish for "change heads", but my printer refuses to use a different extruder.  I'll keep looking into this.

mron, 2017-04-29:
I did some checking. `M108` is ReplicatorG gcode for "set extruder speed". The correct code is either `T[extruder#]` or `M6 T[extruder#]`

mron, 2017-05-01:
After further checking and getting input from the author of gpx, `M108` is correct, but not widely supported. His version does support `M108` however he recommends not using "Sailfish" as a gcode flavor with gpx, and to use "reprap".

I think this is issue is closed.

alexrj, 2017-05-01:
I'm totally lost when it comes to the Makerbot world. Now I don't understand why we have a Sailfish G-code flavor.

closed 2017-05-01 by alexrj