slic3r/Slic3r #3914: Errors building on OS X El Capitan

mron, 2017-04-28:
### Version
Building 1.3

### Operating system type + version
OS X 10.11

### Behavior
Errors building while following "Running Slic3r from git on OS X"
"Installing the stand-alone Command Line Tools" - No Errors
"Installing Perl (via Perlbrew)" - No Errors
"Build Boost" - LATEST BOOST 1.64 - No Errors
"Building Slic3r"
git clone git://github.com/alexrj/Slic3r - No Errors
    cd Slic3r
    BOOST_DIR=~/boost_1_64_0 SLIC3R_STATIC=1 perl Build.PL ----

Several errors like these:

cc -I/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE -arch i386 -arch x86_64 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -xc++ -Isrc/libslic3r -I/Users/ron/boost_1_64_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -U__STRICT_ANSI__ -std=c++11 -stdlib=libc++ -DBOOST_THREAD_DONT_USE_CHRONO -DBOOST_NO_CXX11_RVALUE_REFERENCES -DBOOST_THREAD_USES_MOVE -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -arch i386 -arch x86_64 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -Os -o src/perlglue.o src/perlglue.cpp
In file included from src/perlglue.cpp:2:
In file included from src/xsinit.h:39:
In file included from /System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/perl.h:3476:
/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/pad.h:323:17: error: invalid suffix on literal; C++11 requires a space between literal
      and identifier [-Wreserved-user-defined-literal]
              "Pad 0x%"UVxf"[0x%"UVxf"] set_cur    depth=%d\n", \

There are a total of 8 "error: invalid suffix on literal; C++11 requires a space between literal"

I notice: "In file included from /System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/perl.h"
Should this be looking at the local "~/perl5/..."  I installed during this process?

mron, 2017-04-28:
OOPS -- I think I left out the "Switch to the built perl: perlbrew switch threaded-perl-5.24.1" part...

Trying again ... Sorry

mron, 2017-04-28:
Still have same errors after switch. Here is a typical 'cc' call - 

`cc -I/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE -arch i386 -arch x86_64 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -xc++ -Isrc/libslic3r -I/Users/ron/boost_1_64_0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -U__STRICT_ANSI__ -std=c++11 -stdlib=libc++ -DBOOST_THREAD_DONT_USE_CHRONO -DBOOST_NO_CXX11_RVALUE_REFERENCES -DBOOST_THREAD_USES_MOVE -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -arch i386 -arch x86_64 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -Os -o src/polypartition.o src/polypartition.cpp`

Is it correct to look in "-I/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE"

lordofhyphens, 2017-04-28:
No, it isn't correct. Your build should be using the perlbrew version and not the system perl.

mron, 2017-04-28:
More info:
After running `./b2 link=static runtime-link=static variant=release` this is the output:
`The Boost C++ Libraries were successfully built!

The following directory should be added to compiler include paths:

    /Users/ron/boost_1_64_0

The following directory should be added to linker library paths:

    /Users/ron/boost_1_64_0/stage/lib`

I assumed these path changes and libraries additions are made with the `BOOST_DIR=~/boost_1_63_0` variable and done in the BUILD.pl. If it's now, is there something I need to do?

alexrj, 2017-05-01:
You seem to still haven't issued the `perlbrew switch` command. After issueing it, don't close the terminal as it opens a subshell where you issue the `perl Build.PL` command. Otherwise it will look for the system perl paths instead of the perlbrew one.

alexrj, 2017-05-16:
Well, I'm closing this one as there are no updates.

closed 2017-05-16 by alexrj