slic3r/Slic3r #3917: Post processing not functioning on latest build.

th0mpy, 2017-04-30:
### Version
1.3.0 DEV - slic3r-1.3.0-dev-cc1c9b0.dmg

### Operating system type + version
MAC OS 10.12.4

### Behavior
Post processing does not appear to run. I have a script running which creates the wipe towers for multi-color printing... It works on 1.2.9 but not on the latest. No errors occur, yet, the resultant GCODE does not show to have executed the script. Again, works fine in 1.2.9. 

#### STL/Config (.ZIP) where problem occurs
[Slic3r test files.zip](https://github.com/alexrj/Slic3r/files/966592/Slic3r.test.files.zip)

lordofhyphens, 2017-04-30:
This is probably packaging related.

th0mpy, 2017-04-30:
For testing purposes here is the script file.

[wipe_towers_v02.pl.zip](https://github.com/alexrj/Slic3r/files/966809/wipe_towers_v02.pl.zip)

alexrj, 2017-05-01:
Are you sure the script doesn't get executed? The script might be failing. Maybe it's looking for some syntax which is not used anymore. I just tested, and the script *is* executed but doesn't alter the G-code.
Troubleshooting hints:

* Try setting a non-existent script. Slic3r should warn you.
* Try using another post-processing script (for example the flowrate.pl one available in the Slic3r repository).
* Try launching Slic3r from terminal: `Slic3r.app/Contents/MacOS/Slic3r --gui --debug`; you'll see which scripts are called.

th0mpy, 2017-05-01:
I tried pointing to a non-existent file and I do get an error (file not found essentially).  

I'll test it out when I get home tonight. Since it worked in 1.2.9 do you have an idea of which functions are deprecated in 1.3.0?

alexrj, 2017-05-01:
Functions? The script doesn't call any functions, it just parses the G-code. It should be fixed.

closed 2017-05-01 by alexrj