slic3r/Slic3r #3918: External perimeter speed

h5t4, 2017-04-30:
### Version
1.3.0-dev
$ git rev-parse HEAD
1a788196fada65528fc28ced144d9d5c9537412a

### Operating system type + version
$ uname -o -s -r 
Linux 4.10.11-100.fc24.x86_64 GNU/Linux

### Behavior
#### Expected result:
	External perimeter is printed slower eg. external_perimeter_speed is smaller than perimeter_speed.
#### Actual result:
	Perimeter speed is is same for all perimeters.
 
[external_perimeter_speed_issue.zip](https://github.com/alexrj/Slic3r/files/966819/external_perimeter_speed_issue.zip)

lordofhyphens, 2017-04-30:
This is generally desired behavior. If you want the speed to be the same for both, set it in your config.

lordofhyphens, 2017-04-30:
You are using autospeed, which tries to maintain constant flow. External perimeters tend to be thinner

ForsakenNGS, 2018-02-21:
The external perimiters are always printed at the very end of the layer (or at the very beginning if configured "external perimiters first"), right? Wouldn't it be possible to switch the extrusion speed within the layer once? e.g.:
 - Print everything except the external perimiters with high speed (e.g. 8mm³/s)
 - Print the external perimiters with lower speed for better quality (e.g. 5mm³/s)

This way the extrusion amount would still remain very stable, but improve the surface finish potentially by a lot.