slic3r/Slic3r #3919: Crash with specific STL

multiplemonomials, 2017-05-01:
### Version
automated dev build slic3r-master.2017.04.29.989 (from [here](http://dl.slic3r.org/dev/win/))

### Operating system type + version
Windows 10 Build 1703

### Behavior
* The latest Slic3r build crashes when loading the attached STL into the plater.  It's a kind of screwed up file, I know; I'm still a beginner with Blender.  However, when I use Slic3r 1.2.9 instead of the dev version, it loads and works fine.
* Step to Reproduce: Press `Add...` in the plater window and select the attached STL
  * Background slicing is enabled
* _Expected Results_
  * The file loads and is able to be sliced
* _Actual Results_
  * Slic3r crashes.  Backtrace from MinGW-w64 GDB follows:
```
Program received signal SIGSEGV, Segmentation fault.
0x000000002b9bc359 in boot_Slic3r__XS ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
(gdb) bt
#0  0x000000002b9bc359 in boot_Slic3r__XS ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
#1  0x000000002b9bd412 in boot_Slic3r__XS ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
#2  0x000000002b973991 in boot_Slic3r__XS ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
#3  0x000000002b9c2ac3 in boot_Slic3r__XS ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
#4  0x000000002b870dcf in ?? ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Slic3r\XS\XS.xs.dll
#5  0x000000006db8ecc0 in perl524!Perl_clear_defarray () from C:\Program Files\Slic3r-dev\perl524.dll
#6  0x000000006dbe43e6 in perl524!Perl_runops_standard () from C:\Program Files\Slic3r-dev\perl524.dll
#7  0x000000006db9cdcd in perl524!Perl_call_sv () from C:\Program Files\Slic3r-dev\perl524.dll
#8  0x0000000070353b9f in boot_Wx () from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Wx\Wx.xs.dll
#9  0x00000000047d3106 in wxAppConsoleBase::CallEventHandler(wxEvtHandler*, wxEventFunctor&, wxEvent&) const ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#10 0x0000000004932aee in wxEvtHandler::ProcessEventIfMatchesId(wxEventTableEntryBase const&, wxEvtHandler*, wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#11 0x0000000004933087 in wxEvtHandler::SearchDynamicEventTable(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#12 0x0000000004933382 in wxEvtHandler::TryHereOnly(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#13 0x000000000493341a in wxEvtHandler::ProcessEventLocally(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#14 0x0000000004933498 in wxEvtHandler::ProcessEvent(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#15 0x0000000061cc0a55 in wxWindowBase::TryAfter(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxmsw310u_c
ore_gcc_custom.dll
#16 0x0000000004934d9c in wxEvtHandler::SafelyProcessEvent(wxEvent&) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#17 0x0000000061af5d7b in wxButton::SendClickEvent() ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxmsw310u_-
--Type <return> to continue, or q <return> to quit---
core_gcc_custom.dll
#18 0x0000000061ac23ab in wxWindow::HandleCommand(unsigned short, unsigned short, HWND__*) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxmsw310u_c
ore_gcc_custom.dll
#19 0x0000000061acc955 in wxWindow::MSWHandleMessage(long long*, unsigned int, unsigned long long, long long) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxmsw310u_c
ore_gcc_custom.dll
#20 0x0000000061abc68c in wxWindow::MSWWindowProc(unsigned int, unsigned long long, long long) ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxmsw310u_c
ore_gcc_custom.dll
[... 43 lines cut ...]
#43 0x000000000480dc92 in wxEventLoopManual::ProcessEvents() ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#44 0x000000000480dd11 in wxEventLoopManual::DoRun() ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#45 0x000000000480da64 in wxEventLoopBase::Run() ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#46 0x00000000047d7e9a in wxAppConsoleBase::MainLoop() ()
   from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\Alien\wxWidgets\msw_3_1_0_uni_gcc_3_4\lib\wxbase310u_
gcc_custom.dll
#47 0x000000007041652b in boot_Wx () from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Wx\Wx.xs.dll
#48 0x000000007025dbc4 in ?? () from C:\PROGRA~1\SLIC3R~1\LOCAL-~1\lib\perl5\MSWin32-x64-multi-thread\auto\Wx\Wx.xs.dll
#49 0x000000006db8ecc0 in perl524!Perl_clear_defarray () from C:\Program Files\Slic3r-dev\perl524.dll
#50 0x000000006dbe43e6 in perl524!Perl_runops_standard () from C:\Program Files\Slic3r-dev\perl524.dll
#51 0x000000006dba4994 in perl_run () from C:\Program Files\Slic3r-dev\perl524.dll
#52 0x000000006dbf6f78 in perl524!RunPerl () from C:\Program Files\Slic3r-dev\perl524.dll
#53 0x00000000004019c7 in main ()
```
_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
[Raven Seal STL.zip](https://github.com/alexrj/Slic3r/files/967124/Raven.Seal.STL.zip)

lordofhyphens, 2017-05-01:
Feed the model through service.netfabb.com.

lordofhyphens, 2017-05-01:
In short, GIGO. 95% sure this is in admesh.

multiplemonomials, 2017-05-01:
This does work perfectly in version 1.2.9 though.  Even if it is a messed up STL, I feel like it shouldn't crash the whole program.

alexrj, 2017-05-01:
gdb output:

```
Program received signal EXC_BAD_ACCESS, Could not access memory.
Reason: 13 at address: 0x0000000000000000
0x0000000101212984 in stl_compare_function [inlined] () at /Users/al/Documents/git/Slic3r/xs/src/admesh/connect.c:264
264	  if(edge_a->facet_number == edge_b->facet_number) {
```

alexrj, 2017-05-01:
I agree Slic3r shouldn't crash even with garbage. However I wasn't able to fix this admesh bug, so I'll leave it open.