slic3r/Slic3r #3921: deserializing serialized value for bed_shape

voidloop, 2017-05-01:
### Version
```
% git describe --tag
1.2.9-982-gdff61d1
```
```
% git rev-parse HEAD
dff61d10186f8e0da5961546c544756eae0cf51c
```

### Operating system type + version
```
% cat /etc/redhat-release 
Fedora release 25 (Twenty Five)
```

### Behavior
Hi, I've successfully compiled the master branch of Slic3r repository 

```
% perl Build.PL
% perl Build.PL --gui
```

but the slic3r script doesn't want run! 
After compiling Slic3r this is the error message when I try to run it: 
```
% perl slic3r.pl
Error in function apply_only at src/libslic3r/Config.cpp:269: Unexpected failure when deserializing serialized value for bed_shape
         at /home/exp/my_app/Slic3r/lib/Slic3r/Config.pm line 37.
        Slic3r::Config::new_from_defaults("Slic3r::Config") called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/Plater.pm line 765
        Slic3r::GUI::Plater::config(Slic3r::GUI::Plater=HASH(0x5696a30), 1) called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/Plater.pm line 557
        Slic3r::GUI::Plater::_on_select_preset(Slic3r::GUI::Plater=HASH(0x5696a30), "printer") called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/Plater.pm line 497
        Slic3r::GUI::Plater::new("Slic3r::GUI::Plater", Wx::Notebook=HASH(0x56967f0)) called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/MainFrame.pm line 101
        Slic3r::GUI::MainFrame::_init_tabpanel(Slic3r::GUI::MainFrame=HASH(0x5696598)) called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/MainFrame.pm line 33
        Slic3r::GUI::MainFrame::new("Slic3r::GUI::MainFrame") called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI.pm line 163
        Slic3r::GUI::OnInit(Slic3r::GUI=HASH(0x339fd88)) called at /home/exp/my_app/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/Wx/App.pm line 36
        eval {...} called at /home/exp/my_app/Slic3r/local-lib/lib/perl5/x86_64-linux-thread-multi/Wx/App.pm line 36
        Wx::App::new("Slic3r::GUI") called at slic3r.pl line 116
```

What is the problem?

#### STL/Config (.ZIP) where problem occurs
No STL, no config files.

Thank you.

alexrj, 2017-05-01:
This might be related to the existing configs. Could you please zip the `~/.Slic3r` folder and share it with us?

voidloop, 2017-05-01:
I've also tried to rename the ~/.Slic3r directory thinking that error message would be disappeared but it didn't. It's the same error message of my previous comment.

My config files: [slic3r-configs.tar.gz](https://github.com/alexrj/Slic3r/files/968190/slic3r-configs.tar.gz)

```
% mv ~/.Slic3r ~/slic3r-configs
% perl slic3r.pl 
Error in function apply_only at src/libslic3r/Config.cpp:269: Unexpected failure when deserializing serialized value for bed_shape
         at /home/exp/my_app/Slic3r/lib/Slic3r/Config.pm line 37.
        Slic3r::Config::new_from_defaults("Slic3r::Config") called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/Plater.pm line 765
...
```
Thank you.

alexrj, 2017-05-01:
This is very interesting, and I can't reproduce it.
What Boost version are you using?
What gcc version?

I just pushed a minor change to a unit test. Could you run `cd xs ; ./Build test` and confirm all the tests run with no errors?

voidloop, 2017-05-01:
Ok.  I confirm all the tests run with no errors. 

```
% cd xs ; ./Build test

...compiling...

t/01_trianglemesh.t ............. ok     
t/03_point.t .................... ok     
t/04_expolygon.t ................ ok     
t/05_surface.t .................. ok     
t/06_polygon.t .................. ok     
t/07_extrusionpath.t ............ ok   
t/08_extrusionloop.t ............ ok     
t/09_polyline.t ................. ok     
t/10_line.t ..................... ok     
t/11_clipper.t .................. ok     
t/12_extrusionpathcollection.t .. ok     
t/13_polylinecollection.t ....... ok   
t/14_geometry.t ................. ok   
t/15_config.t ................... ok       
t/16_flow.t ..................... ok   
t/17_boundingbox.t .............. ok   
t/18_motionplanner.t ............ ok     
t/19_model.t .................... ok   
t/20_print.t .................... ok   
t/21_gcode.t .................... ok   
t/22_exception.t ................ ok   
All tests successful.
Files=21, Tests=507,  2 wallclock secs ( 0.13 usr  0.05 sys +  1.86 cusr  0.30 csys =  2.34 CPU)
Result: PASS
```
The version of the software you asked me:

```
% gcc --version
gcc (GCC) 6.3.1 20161221 (Red Hat 6.3.1-1)
```
```
% rpm -qa | grep boost
boost-test-1.60.0-10.fc25.x86_64
boost-coroutine-1.60.0-10.fc25.x86_64
boost-container-1.60.0-10.fc25.x86_64
boost-context-1.60.0-10.fc25.x86_64
boost-python-1.60.0-10.fc25.x86_64
boost-signals-1.60.0-10.fc25.x86_64
boost-timer-1.60.0-10.fc25.x86_64
boost-devel-1.60.0-10.fc25.x86_64
boost-chrono-1.60.0-10.fc25.x86_64
boost-filesystem-1.60.0-10.fc25.x86_64
boost-thread-1.60.0-10.fc25.x86_64
boost-program-options-1.60.0-10.fc25.x86_64
boost-atomic-1.60.0-10.fc25.x86_64
boost-graph-1.60.0-10.fc25.x86_64
boost-serialization-1.60.0-10.fc25.x86_64
boost-date-time-1.60.0-10.fc25.x86_64
boost-wave-1.60.0-10.fc25.x86_64
boost-system-1.60.0-10.fc25.x86_64
boost-regex-1.60.0-10.fc25.x86_64
boost-iostreams-1.60.0-10.fc25.x86_64
boost-1.60.0-10.fc25.x86_64
boost-random-1.60.0-10.fc25.x86_64
boost-locale-1.60.0-10.fc25.x86_64
boost-math-1.60.0-10.fc25.x86_64
boost-type_erasure-1.60.0-10.fc25.x86_64
boost-log-1.60.0-10.fc25.x86_64
```
What can I try?

Thank you.

alexrj, 2017-05-02:
Could you please replace the last function in xs/src/libslic3r/Config.cpp with the following code and recompile? Then launch Slic3r and post here the console output.

```
bool
ConfigOptionPoints::deserialize(std::string str, bool append) {
	std::cout << "Deserializing: " << str << std::endl;
	if (!append) this->values.clear();

	std::vector<std::string> tokens;
	boost::split(tokens, str, boost::is_any_of("x,"));
	std::cout << "Got " << tokens.size() << " tokens:" << std::endl;
	for (auto s : tokens) std::cout << "* " << s << std::endl;
	if (tokens.size() % 2) return false;

	try {
		for (size_t i = 0; i < tokens.size(); ++i) {
			Pointf point;
			point.x = boost::lexical_cast<coordf_t>(tokens[i]);
			point.y = boost::lexical_cast<coordf_t>(tokens[++i]);
			this->values.push_back(point);
		}
	} catch (boost::bad_lexical_cast &e) {
		printf("%s\n", e.what());
		return false;
	}

	return true;
}
```

voidloop, 2017-05-02:
I removed ./Slic3r directory before run `perl slic3r.pl`.
This is the output: 

```
% rm -rfv ~/.Slic3r -rfv
removed directory '/home/exp/.Slic3r/printer'
removed '/home/exp/.Slic3r/slic3r.ini'
removed directory '/home/exp/.Slic3r/print'
removed directory '/home/exp/.Slic3r/filament'
removed directory '/home/exp/.Slic3r'

% perl slic3r.pl
Deserializing: 
Got 1 tokens:
* 
Error in function apply_only at src/libslic3r/Config.cpp:269: Unexpected failure when deserializing serialized value for bed_shape
         at /home/exp/my_app/Slic3r/lib/Slic3r/Config.pm line 37.
        Slic3r::Config::new_from_defaults("Slic3r::Config") called at /home/exp/my_app/Slic3r/lib/Slic3r/GUI/Plater.pm line 765
...
```
What can I do to help you? could I try another version of boost?
Thank you.

alexrj, 2017-05-03:
This is really really strange.
Are you using the system perl or did you install one via perlbrew?

Any chance you can come in #slic3r on FreeNode IRC and we can have an interactive debugging session?

voidloop, 2017-05-04:
System Perl. 
Ok, I will stay connected in #slic3r on FreeNode.

lordofhyphens, 2017-05-04:
Yeah, you don't want system perl on OSX.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

alexrj, 2017-05-04:
@voidloop, system perl on OS X is not guaranteed to work. Use perlbrew.
Please follow the instructions at https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-OS-X

voidloop, 2017-05-05:
Hey, I'm using Fedora 25. 
But if you want I could try with perlbrew the same.

alexrj, 2017-05-05:
Oh sorry. I mixed up two issues. Well, perlbrew is recommended as well.
It would be nice to understand what's going on with your system perl, but it's hard to do it remotely.

lordofhyphens, 2017-09-07:
Reopening while I check tests around bed_shape specifically (I didn't see any tests having to do with it).

blubaron, 2017-09-19:
I am using Fedora 24 and I have this issue as well.

To get around it I changed the last function in Config.cpp to:

    bool
    ConfigOptionPoints::deserialize(std::string str, bool append) {
	if (!append) this->values.clear();

	std::vector<std::string> tokens;
	boost::split(tokens, str, boost::is_any_of("x,"));
	if ((tokens.size() == 1) && (tokens[0].length() == 0)) {
		std::cout << "Point token 0 size: 0 for:\n";
		Pointf point;
		point.y = point.x = 0;
		this->values.push_back(point);
		return true;
	}
	if (tokens.size() % 2) return false;

	try {
		for (size_t i = 0; i < tokens.size(); ++i) {
			Pointf point;
			point.x = boost::lexical_cast<coordf_t>(tokens[i]);
			point.y = boost::lexical_cast<coordf_t>(tokens[++i]);
			this->values.push_back(point);
		}
	} catch (boost::bad_lexical_cast &e) {
		printf("%s\n", e.what());
		return false;
	}

	return true;
    }

Also, I changed the area around line 267 in Config.cpp to:

        // not the most efficient way, but easier than casting pointers to subclasses
        bool res = my_opt->deserialize( other.option(opt_key)->serialize() );
        if (!res) {
            std::string error = "Unexpected fail when deserializing serialized value for " + opt_key + " Value: " + other.option(opt_key)->serialize() + ".\n";
            CONFESS(error.c_str());
        } else {
            std::cout << " deserializing key: "  + opt_key + " text: " + other.option(opt_key)->serialize() + "\n";
        }

The relavent lines in the output are:
Point token 0 size: 0 for:
 deserializing key: bed_shape text: 

so the option text is empty when it gets to that point.

It is the same for extruder offset:
 deserializing key: external_perimeter_extrusion_width text: 0
 deserializing key: external_perimeter_speed text: 50%
 deserializing key: external_perimeters_first text: 0
 deserializing key: extra_perimeters text: 1
 deserializing key: extruder_clearance_height text: 20
 deserializing key: extruder_clearance_radius text: 20
Point token 0 size: 0 for:
 deserializing key: extruder_offset text: 
 deserializing key: extrusion_axis text: E
 deserializing key: extrusion_multiplier text: 1
 deserializing key: extrusion_width text: 0

Also, when using this output there are 4476 lines in the terminal, so the options are probably being set too much.

I have not seen where the default settings are loaded from, so this is as much information as I can give.

lordofhyphens, 2017-09-19:
@blubaron the defaults are read straight out of PrintConfig.cpp (they're compiled in).

lordofhyphens, 2017-09-19:
It looks like in that instance the object is empty from my reading of ConfigOptionPoints::serialize() (Config.hpp, around line 383).

Why? I have no idea. 

For a test, try replacing the contents of t/config.t with the following:
````
use Test::More tests => 3;
use strict;
use warnings;

BEGIN {
    use FindBin;
    use lib "$FindBin::Bin/../lib";
    use local::lib "$FindBin::Bin/../local-lib";
}

use Slic3r;
use Slic3r::Test;

{
    my $config = Slic3r::Config->new_from_defaults;
    $config->set('perimeter_extrusion_width', '250%');
    ok $config->validate, 'percent extrusion width is validated';
    my @keys = ('bed_shape');
    my $config2 = Slic3r::Config->new;
    $config2->apply_only($config, \@keys);
    my $z = $config2->serialize('bed_shape');
    is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
    $z = $config->serialize('bed_shape');
    is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
}

__END__
````

This should trigger the problem on demand on your system. 

@blubaron I'll also need to know the compiler version you're using with Slic3r, boost, etc.

lordofhyphens, 2017-09-19:
@blubaron Also it would be helpful to know if either of the precompiled builds break on your system.

blubaron, 2017-09-19:
hmm it may be a compiler issue or something is not being allocated in the right order on fedora.

because inserting:
        (void) opt->serialize();
to PrintConfig.cpp line 42 makes the issue go away, with the later strings for bed_shape and extruder offest show/have the right values.

Also I do have the same compiler as above.
gcc (GCC) 6.3.1 20161221 (Red Hat 6.3.1-1)

Also the prebuilt slic3r-1.3.0-dev-1a2df5e-linux-x64 works for me.

lordofhyphens, 2017-09-19:
@blubaron Does the modified test pass or fail for you?

blubaron, 2017-09-19:
Is there a way to run the test without the full build process? I am not familiar with perl.

lordofhyphens, 2017-09-19:
````gcc version 6.3.0 20170516 (Debian 6.3.0-18)```` is my compiler. The build server is using 4.9. 

If you set local::lib appropriately with ````eval $(perl -Mlocal::lib=/path/to/slic3r/local-lib)```` you can use ````prove t/config.t```` to run the script. You can also just run the tests with perl directly (they're just perl scripts after all).

blubaron, 2017-09-19:
The test passes whether the above line is in there or not. (whether I get the error or not.)

$ perl t/config.t
1..3
ok 1 - percent extrusion width is validated
ok 2 - running apply_only
ok 3 - running apply_only

lordofhyphens, 2017-09-19:
What about 
````
{
    my $config = Slic3r::Config->new_from_defaults;
    $config->set('perimeter_extrusion_width', '250%');
    ok $config->validate, 'percent extrusion width is validated';
    $config = Slic3r::Config->new;
    $config->apply_static(Slic3r::Config::Full->new);
    my $z = $config->serialize('bed_shape');
    is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
}
````

for the tests? (set the number of tests at the top to 2).

lordofhyphens, 2017-09-19:
If you're getting the same error as the original report, then the backtrace means that apply_static() is getting called (which is what that second test item does), which should avoid any local environment stuff from possibly happening.

I'm very hesitant to change any SW if we don't have a test to catch regressions, especially for something that might just be over-aggressive optimizations.

Slic3r's build environment pays attention to CFLAGS environment variable. What's the contents of that environment var?

blubaron, 2017-09-19:
The tests still pass.

$ prove t/config.t
t/config.t .. ok   
All tests successful.
Files=1, Tests=2,  1 wallclock secs ( 0.05 usr  0.01 sys +  0.35 cusr  0.05 csys =  0.46 CPU)
Result: PASS
$ perl t/config.t
1..2
ok 1 - percent extrusion width is validated
ok 2 - running apply_only

I do not have the CFLAGS environment variable.

I agree that the above line should not be added directly to the code. It is an odd line in an odd place. It just provides more information for finding the problem.

lordofhyphens, 2017-09-19:
I'm basically out of ideas here, especially since I cannot reproduce this issue on anything that I build.

lordofhyphens, 2017-09-19:
Those tests are run without your workaround, correct?

blubaron, 2017-09-19:
I changed the tests to new_from_defaults and it fails:

{
    my $config = Slic3r::Config->new_from_defaults;
    $config->set('perimeter_extrusion_width', '250%');
    ok $config->validate, 'percent extrusion width is validated';
    $config = Slic3r::Config->new;
    $config->new_from_defaults(Slic3r::Config::Full->new);
    my $z = $config->serialize('bed_shape');
    is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
}

$ perl t/config.t
1..2
ok 1 - percent extrusion width is validated
Segmentation fault (core dumped)

$ prove t/config.t
t/config.t .. Failed 1/2 subtests 

Test Summary Report
-------------------
t/config.t (Wstat: 139 Tests: 1 Failed: 0)
  Non-zero wait status: 139
  Parse errors: Bad plan.  You planned 2 tests but ran 1.
Files=1, Tests=1,  1 wallclock secs ( 0.06 usr  0.00 sys +  0.31 cusr  0.05 csys =  0.42 CPU)
Result: FAIL

blubaron, 2017-09-19:
yes, without the workaround.

Also, even with the workaround, the new_from_defaults test fails.

lordofhyphens, 2017-09-19:
I don't believe you are using the function properly there; new_from_defaults is a constructor. I don't think it's proper to call new_from_defaults on an object.

lordofhyphens, 2017-09-19:
The reason it's segfaulting is that it's trying to guess the name of the class (Slic3r::Config) but when you call it like `$config->new_from_defaults;`, the `$class` variable ends up being `Slic3r::Config=SCALAR(0x55847fe24cf8)`, which then turns into `Slic3r::Config=SCALAR(0x55847fe24cf8)->new`, which doesn't exist, instead of `Slic3r::Config->new`.

blubaron, 2017-09-19:
Yeah I wasn't.
with the test:
{
    my $config = Slic3r::Config->new_from_defaults;
    my $z = $config->serialize('bed_shape');
    is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
}

passes

blubaron, 2017-09-19:
Is there a way to start slicer from the end of the test? So the same perl environment as the test was run in?

blubaron, 2017-09-19:
hmm with the beginning of slic3r.pl:

    #!/usr/bin/env perl
    use Test::More tests => 1;
    
    use strict;
    use warnings;
    
    BEGIN {
        use FindBin;
        use lib "$FindBin::Bin/lib";
        use local::lib '--no-create', "$FindBin::Bin/local-lib";
    }
    
    use Slic3r;
    use Slic3r::Test;
    
    {
        my $config = Slic3r::Config->new_from_defaults;
        my $z = $config->serialize('bed_shape');
        is $z, "0x0,200x0,200x200,0x200", 'running apply_only';
    }
    
    use File::Basename qw(basename);
    use Getopt::Long qw(:config no_auto_abbrev);
    ...

it starts without the error.

blubaron, 2017-09-19:
actually it starts without error with:

{
    my $config = Slic3r::Config->new_from_defaults;
    my $z = $config->serialize('bed_shape');
}

anywhere before:

    $gui = Slic3r::GUI->new;

of line 118 of slic3r.pl.

lordofhyphens, 2017-09-19:
Not that I am aware.

lordofhyphens, 2017-09-19:
That is really weird.

blubaron, 2017-09-19:
It is weird and it gets worse.

the problem can be narrowed down to putting:

	my $z = Slic3r::Config::Full->new->serialize('bed_shape');

before:

    my $self = $class->SUPER::new($parent, -1, Wx::wxDefaultPosition, Wx::wxDefaultSize, 0, "", $attrib);

around line 94 of ./lib/Slic3r/GUI/3DScene.pm. After this line the error is given.

After putting in some print statements:

    # we request a depth buffer explicitely because it looks like it's not created by 
    # default on Linux, causing transparency issues
    my $z;
    #$z = Slic3r::Config::Full->new->serialize('bed_shape');
    #print $z; print ".\n";
    
    my $self = $class->SUPER::new($parent, -1, Wx::wxDefaultPosition, Wx::wxDefaultSize, 0, "", $attrib);
    
    $z = Slic3r::Config::Full->new->serialize('bed_shape');
    print $z; print ".\n";

I get the output:

```
$ perl slic3r.pl --gui
test1
test2
Slic3r::GUI::Plater::3D::new
Slic3r::GUI::3DScene::Base::new
.
end Slic3r::GUI::3DScene::Base::new
end Slic3r::GUI::Plater::3D::new
Slic3r::GUI::3DScene::Base::new
.
end Slic3r::GUI::3DScene::Base::new
new_from_defaults
Error in function apply_only at src/libslic3r/Config.cpp:270: Unexpected fail when deserializing serialized value for bed_shape Value: .

	 at /home/liveuser/Slic3r-master/lib/Slic3r/Config.pm line 42.
	Slic3r::Config::new_from_defaults("Slic3r::Config") called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI/Plater.pm line 848
	Slic3r::GUI::Plater::config(Slic3r::GUI::Plater=HASH(0x3b13e68), 1) called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI/Plater.pm line 617
	Slic3r::GUI::Plater::_on_select_preset(Slic3r::GUI::Plater=HASH(0x3b13e68), "printer") called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI/Plater.pm line 549
	Slic3r::GUI::Plater::new("Slic3r::GUI::Plater", Wx::AuiNotebook=HASH(0x3990818)) called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI/MainFrame.pm line 117
	Slic3r::GUI::MainFrame::_init_tabpanel(Slic3r::GUI::MainFrame=HASH(0x3ac4d68)) called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI/MainFrame.pm line 35
	Slic3r::GUI::MainFrame::new("Slic3r::GUI::MainFrame") called at /home/liveuser/Slic3r-master/lib/Slic3r/GUI.pm line 167
	Slic3r::GUI::OnInit(Slic3r::GUI=HASH(0x17fa8c0)) called at /home/liveuser/Slic3r-master/local-lib/lib/perl5/x86_64-linux-thread-multi/Wx/App.pm line 36
	eval {...} called at /home/liveuser/Slic3r-master/local-lib/lib/perl5/x86_64-linux-thread-multi/Wx/App.pm line 36
	Wx::App::new("Slic3r::GUI") called at slic3r.pl line 130

```
with the first lines uncommented, I get:

```
$ perl slic3r.pl --gui
test1
test2
Slic3r::GUI::Plater::3D::new
Slic3r::GUI::3DScene::Base::new
0x0,200x0,200x200,0x200.
0x0,200x0,200x200,0x200.
end Slic3r::GUI::3DScene::Base::new
end Slic3r::GUI::Plater::3D::new
Slic3r::GUI::3DScene::Base::new
0x0,200x0,200x200,0x200.
0x0,200x0,200x200,0x200.
end Slic3r::GUI::3DScene::Base::new
new_from_defaults
new_from_defaults
new_from_defaults
new_from_defaults
new_from_defaults
new_from_defaults
new_from_defaults
new_from_defaults
```

and it starts correctly.

I cannot narrow it down any further, because the base is Wx::GLCanvas? and I could not find it.

It seems the problem depends on when the first ConfigOptionPoints::serialize() is called?
I wonder if ConfigOptionBools::serialize() will show the same problem, but I have no idea how to test it.

lordofhyphens, 2017-09-20:
Wx::GlCanvas is part of wxperl, which is unfortunately unmaintained.

blubaron, 2017-09-22:
Is there anything else I should try?

lordofhyphens, 2017-09-22:
Nothing comes to mind, sorry.