slic3r/Slic3r #3922: Slicer Does not finish gcode file

Louis-Cypher, 2017-05-02:
### Version
1.3.0 dev

### Operating system type + version
Windows 7 64bit

### Behavior
* Export of gcode stops. Slic3r says export finished (see screenshot) but in Windows explorer there is only a gcode.temp file visible

* Slice file and try to export gcode

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

[image]

I sadly cannot upload the ANY zip file.  Hence only, the config called config.txt
[config.txt](https://github.com/alexrj/Slic3r/files/970248/config.txt)
. The stl is the one from thingiverse ( http://www.thingiverse.com/thing:730263 ). Please scale it to 91%

Best,

Markus

Louis-Cypher, 2017-05-02:
The most current build (today) even doesn't read the stl file.

foreachthing, 2017-05-02:
You cannot have öäü in your path or filename!
I just had the same issue today! :-)

Louis-Cypher, 2017-05-02:
But I was able to open that file with the build from a couple of weeks ago yesterday (and still am). Isn't this a bug?

foreachthing, 2017-05-02:
@Louis-Cypher that's what I thought, too! I'm glad you brought it up.
It's no major bug and easy to work around but pretty annoying for us with öäü on our keyboards.

Louis-Cypher, 2017-05-02:
The problem doesn't occur in the current build as soon as I remove the "ä" from the path.

closed 2017-05-02 by Louis-Cypher

alexrj, 2017-05-02:
That bug is #2955

Louis-Cypher, 2017-05-02:
I don't think so:

Shouldn't loaded config be saved automatically ? #2995