slic3r/Slic3r #3923: Slic3r doesn't open files in paths with "ä"

Louis-Cypher, 2017-05-02:
### Version
_Version of Slic3r used goes here_

1.3.0 dev buildbot 2.5.2017
### Operating system type + version
_Win 7 64it_

### Behavior
* Slic3r doesn't open files in paths with "ä". It complains about not being able to read the STL (which is not true; it can be opened in a different path)
* _Steps needed to reproduce the problem_
  * generate a path having an ä
   * Try to open a stlfile in path
[image]

_
#### STL/Config (.ZIP) where problem occurs
Not related to any config

lordofhyphens, 2017-05-03:
#3763

Dupe (same underlying problem, exacerbated by how Windows does things).

closed 2017-05-03 by lordofhyphens

lordofhyphens, 2017-05-18:
@Louis-Cypher Recommend following #2955

alexrj, 2017-05-19:
This was fixed now. It can be tested with the latest dev build available at http://dl.slic3r.org/dev/win