slic3r/Slic3r #3926: GUI Spanish translation

mekod4, 2017-05-03:
### Version
1.2.9 (8c77f4763854d33255d61f7a7336c229dad080d7)

### Operating system type + version
Windows 10

### Behavior
Is there any documentation for the language translation process? I would like to have a Spanish GUI.

I've took a glipse at the code, and seems that it will consist in changing the GUI data across the GUI Perl modules. Am I missing something? Any recommendations?

Thanks for the help, and for the great tool you've created!

lordofhyphens, 2017-05-03:
The current effort is collecting translations for the options, see the i18n
branch.

Currently does not have an effect, need to do more work on the C++ UI first.

lordofhyphens, 2017-05-03:
Please don't fork Slic3r to change all the strings like that. We are going to use .po fIles like other sane people do ;)

mekod4, 2017-05-03:
Great :D Do you have an estimate about how much effort is needed to start adopting i18n?

lordofhyphens, 2017-05-03:
Lots.

There is the translation itself and the porting of the GUI.

Most of the work is straightforward. I have been working on one of the
trickier parts.