slic3r/Slic3r #3928: Crash during slicing (generating infill)

VanessaE, 2017-05-03:
### Version

git commit c82302e284b08b8ba43e7dc641cfd386418f3561

### Operating system type + version

Debian sid

### Behavior

Slic3r Loads the attached STL just fine, and begins slicing without issue.  It gets through the "Processing triangulated mesh" stage after quite some time, then eventually moves on to "Infilling layers".  Not too long after it starts that stage (several minutes), it dies with segfault.  I managed to get a core dump (with help from @lordofhyphens to load it into gdb).  Here's the initial output, but I can't put the full backtrace here because it's too big for github (over 64kB):

```
(gdb) core-file /home/vanessa/slic3r.core
[New LWP 25427]
[New LWP 23712]
[New LWP 25422]
[New LWP 25423]
[New LWP 25424]
[New LWP 25425]
[New LWP 23708]
[New LWP 23713]
[New LWP 23709]
[New LWP 23710]
[New LWP 23711]
[New LWP 23714]
[New LWP 23715]
[New LWP 23784]
[New LWP 25426]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `perl ./slic3r.pl'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  std::vector<Slic3r::Point, std::allocator<Slic3r::Point> >::~vector (
    this=<optimized out>, __in_chrg=<optimized out>)
    at /usr/include/c++/6/bits/stl_vector.h:426
426	      { std::_Destroy(this->_M_impl._M_start, this->_M_impl._M_finish,
[Current thread is 1 (Thread 0x7f96a08e5700 (LWP 25427))]
```

#### STL/Config (.ZIP) where problem occurs

[EiffelTower_fixed_split_300-percent.stl.zip](https://github.com/alexrj/Slic3r/files/973135/EiffelTower_fixed_split_300-percent.stl.zip)
[segfault-config.ini.zip](https://github.com/alexrj/Slic3r/files/973134/segfault-config.ini.zip)

The model is originally from http://www.thingiverse.com/thing:912478 ... except that I used Slic3r to scale it up (300%), "Cut" it into pieces, arrange the parts to fit my plate, and export the plate to the above STL.

VanessaE, 2017-05-03:
Here's the output of `thread apply all bt full`:

[slic3r-dump.log.zip](https://github.com/alexrj/Slic3r/files/973190/slic3r-dump.log.zip)
(~2.2MB, expands to ~35MB of log text)

0xc0ffea, 2017-05-03:
Same issue with another eiffel tower (attached)
[Eiffel_Tower_199.5mm_tall_with_nearly_all_detail_printable_at_0.4mm_nozzle.zip](https://github.com/alexrj/Slic3r/files/974678/Eiffel_Tower_199.5mm_tall_with_nearly_all_detail_printable_at_0.4mm_nozzle.zip)

alexrj, 2017-05-05:
Looks like we have a problem with Eiffel towers.

alexrj, 2017-05-08:
Thank you for this, I fixed it. Too random to write a regression test, but should be regression-safe.

closed 2017-05-08 by alexrj